iNavFlight/inav #6021: F35 furious fpv board r9mm s-port not working 2.5.x

tecnoadsl, 2020-08-05:
After upgrading from 2.4 to 2.50 s-port is not more working. Tested for inverted and univerted pin receiver side. After downgrade it stars to run again normally on s-port pin.

upeter, 2020-08-12:
> After upgrading from 2.4 to 2.50 s-port is not more working. Tested for inverted and univerted pin receiver side. After downgrade it stars to run again normally on s-port pin.

Same problem here with identical setup, though I could never get telemetry (besides rssi, rxbt) to work.
I use: Open tx 2.3.9
R9M 2019 Flex firmware
Inav 2.5.2

I tried all telemetry_ configurations, like in: https://github.com/iNavFlight/inav/issues/5902 but no luck so far.

My s-port connection is soldered to the inverted s-port pin (number 5) and not s-port/f-port (number 2) of the R9MM receiver. Do you also can confirm that?

tecnoadsl, 2020-08-12:
You should use s-port pin n.2 receiver side and use "set telemetry_halfduplex = OFF". In this way it works again.

upeter, 2020-08-14:
That`s indeed the trick, figured it out myself just to realize you mentionned the solution here ;-). Thanx.

closed 2020-11-30 by DzikuVx