iNavFlight/inav #6025: Let INAV figure out rates on FW

avs2, 2020-08-07:
## Current Behavior

The PIFF controller on fixed wing uses the airplane's maximum roll, pitch and yaw rates for it's stabilization. Finding the correct rates can be a bit of a pain, especially for planes that can't do nice clean rolls and loops without a lot of other pilot input. Measuring the yaw rate is even more difficult, since rudder often has a strong roll response. In that case a black box log is the only method to figure out the rates, but this requires a computer which is often impractical.

## Desired Behavior

The gyroscope should be able to measure the rotation rates on full control surface deflection.

## Suggested Solution

Make `AUTOTUNE` a two-stage process where it first determines the rates on full control surface deflection, and then tunes the PIFF values using those rates. Ideally it would be possible to also do each stage separately if the user only want to do one or the other.

## Who does this impact? Who is this for?

All fixed wing pilots.

## Additional context

avs2, 2020-09-22:
Anyone?

FPVZaphod, 2020-10-01:
Great idea!
Measurement can be initiated once the stick reaches maximum deflection, then wait a short while till the rate reaches a stable value.

mf0o, 2021-01-19:
I really wonder why this hasnt been a topic for a longer time but I guess that setting is underestimated by most pilots. Secondly, with the scale-down-trend more and more smaller FCs dont even have blackbox capabilities - at least my last two planes doesnt.

An AUTOROLL feature would be awesome, extracting rates from DVR or external footage still feels like a nasty workaround. 

How could the function work in detail?
- new mode? (i.e. place on the manual channel)
- watch the roll/pitch channel
- start measuring once the channel is (and keep) at max
- stop measuring after 1sec
- print °deg on status/warning field (no auto-set/save for first release)
- repeat at #2 

Thoughts?

PR #6681 (2021-05-08): FW autotune improvements
_Now tested and working_

This PR improves the autotune mode on fixed wing by letting autotune set the correct rates for your airplane (closes #6025). There are three autotune 'modes': with `AUTO` autotune will automatically adjust the max rate settings according to the capabilities of the airplane by setting the rates consistent with 80% control surface deflection. This leaves 20% room for P and I to provide stabilization. `LIMIT` works the same as `AUTO` but the rates are never increased above the starting values. This could be useful if you have a very agile plane a [... cut]
closed 2021-05-08 by avsaase