iNavFlight/inav #6033: BMP280 on MATEKF405 fails to initialize

CecilFF4, 2020-08-10:
## Current Behavior
Barometer will not always initialize, which causes a hardware error flag and prevents arming.

## Steps to Reproduce
Error steps:
1. Plug in USB to FC by itself or before battery.
2. Baro status remains red (unavailable).
Randomly working steps:
1. Plug in battery before USB.
2. Randomly, baro status will remain red.

## Expected behavior
Any version of iNAV after version 2.1.0 show the erroneous behavior. On iNAV version 2.1.0 or earlier, the baro would initialize (status OK) on USB power alone. This more consistent initialization seems to lead to successful arming when on battery alone.
In addition, and I'm not sure if this is related, but on 2.1.0 and earlier, I had no I2C errors. After that version, I had at least 50.

## Suggested solution(s)
Not sure, but it does seem like the firmware is to blame and not the flight controller because the behavior will change depending on the iNAV version. Also, not sure if relevant, but the baro always initializes on USB power alone in betaflight.

## Additional context
I have full-erased and re-flashed numerous times. The problem persists on the versions mentioned previously (including the latest 2.5.2). Also tried disabling and re-enabling the barometer, with no change.
Dumps:
[2.1.0](https://pastebin.com/zh2QFjq3)
[2.2.0](https://pastebin.com/7HbquAgF)
[2.5.2](https://pastebin.com/zxyDjgr3)

---
- FC Board name and vendor: MATEKF405
- INAV version string: 2.1.0, 2.2.0, 2.5.2

Clawzy, 2020-08-11:
To confirm, I have the same issue with the F405-CTR and version 2.5.2.
# INAV/MATEKF405 2.5.2 Aug  4 2020 / 10:15:56 (faaedc7cb)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

craigmunday, 2020-08-11:
I also have a similar issue, although the baro seems to always initialize if I plug the battery in first.

MATEK F405-CTR
INAV/MATEKF405 2.5.2

CecilFF4, 2020-08-14:
I have an update regarding this issue. Recently, I disassembled the quad to perform unrelated maintenance. Once everything was unplugged from the FC, I connected it to the configurator and the baro status was OK. I checked this numerous times and it remained OK each time. I plugged in all my components then connected the FC to my computer and checked again; the baro status was unavailable consistently. I plugged in one component at a time until the issue presented itself and noted that the gps/compass was the component that caused this to happen.

Again, this issue only presents itself on inav 2.2.0 or later (including 2.5.2). Also, I've attached some pictures to show the gps/compass I'm using (BN880) and how it was soldered to the FC. I checked for continuity between each wire/pin on both the FC and in the gps/compass itself (all open). I've also redumped the inav settings for both situations (with and without the gps/compass).
[image]
[image]

[dump_2.5.2_gps_comp_attached](https://pastebin.com/M4h7XEfK)
[dump_2.5.2_gps_comp_not_attached](https://pastebin.com/k2dfnRzV)

Clawzy, 2020-08-15:
After some additional work and a process of elimination, I resolved the issue I had with this initialisation. It is the GPS setup that causes the problem. The battery needs to be connected from the get go, before connecting the iNav Configurator. Keep the battery connected through the process. Reset the FC setup in the Configurator as the first step and follow these guides exactly.

GPS Video steps: https://youtu.be/POAhNV4-YVg (Thanks Lee)
GPS Setup: https://github.com/iNavFlight/inav/wiki/GPS--and-Compass-setup

CecilFF4, 2020-08-15:
Glad that worked for you. Unfortunately, as I described in the "randomly working" steps, plugging in the battery (first or by itself) does not consistently cause the baro to initialize in my setup. Also, both betaflight and inav 2.1.0 (and earlier) initialize the baro with USB power only. Lastly, the baro can be initialized consistently on USB power only if the GPS/compass is disconnected. So it seems clear that something was introduced in the 2.2.0+ firmwares (maybe with the way inav deals with data on the I2C bus) that causes a miscommunication with the baro when the GPS is present (and even unpowered).

closed 2022-02-10 by DzikuVx