iNavFlight/inav #6036: More Adjustments lines (or alternative system)

avs2, 2020-08-11:
## Current Behavior

There are twelve lines for adjustment ranges that can be configured. On my transmitter (OpenTX), I have setup a momentary switch that cycles through a channel from -100 (1000us) to +100 (2000us) in steps of 10 (50us), each mapped to a different adjustment. Combined with an separate OSD screen with all the adjustable values, this provides an easy way to change many settings in-flight. However, I am running into the limit that only twelve are available.

[image]

## Suggested Solution

Make available more lines on the adjustments tab. This would also require the channel range values to be configurable more precisely than the current 25us (if you want to enable all adjustments using only one channel, like me).

Alternatively, and this would be much easier for the user, the adjustments system can be changed so that you can cycle through a list of adjustments using channel high and low values. 

## Who does this impact? Who is this for?

Everyone that uses adjustments in combinations with the OSD elements. Pilots that use adjustments without the OSD elements would not be helpend by my second solution.

## Additional context

avs2, 2020-08-11:
I see this issue is a duplicate of #5984. I think my alternative suggested solution would still be a useful feature.

b14ckyy, 2020-08-12:
This is exactly what my request said and it is already implemented for 2.6. 
With the update we will have 20 adjustements. I already test this with a beta build and 20 should be more than enough. Othweise the OSD will just be totally filled up. 

In Theory you can already have 40 selectable slots with 25us steps but then you might run into issues with channel outputs as for example s.bus goes from 988 to 2011 by default. iBus from 1000 to 2000 and if someone uses CPPM it can even fluctuate.

closed 2021-02-23 by avsaase