iNavFlight/inav #6045: OSD Statistics Screen has too many lines for NTSC

danarrib, 2020-08-13:
## Current Behavior

The Statistics screen that is displayed after each disarm has too many information lines, more than the NTSC mode is capable to display.

[image]

As a result, some important information is not displayed since it doesn't fit on the screen.

## Steps to Reproduce
1. Use a NTSC camera and set OSD to NTSC
2. Use an aircraft with GPS and Current sensor
3. Arm and fly the aircraft for some time
4. When disarming, pay attention to the statistics screen. The last line should be "DISARMED BY: SWITCH", but it will not be displayed since it doesn't fit on the screen.

## Expected behavior
All the information should be displayed in the screen.

## Suggested solution(s)
Discussing on Telegram, @shellixyz suggested to break the statistics screen in two columns.
Each column should take 13 chars, allowing 2 separation chars between them.
Each label should take 6 chars and each value another 6 chars, with 1 char separating the label from the value.

[image]

This solution would work for most of the current elements of the statistics screen. Only the "Fly Time" element would need some special attention since it currently takes 8 chars.

My suggestion was to make it show time using MM:SS if the flight time is less than one hour, or HH:MM if it takes more than one hour. Or use "H", "M" and "S" chars to make it clear what was the flight time. Like this:
* 24M35S (24 minutes and 35 seconds)
* 02H03M (2 hours and 3 minutes)

@stronnag suggested to use "MMM:SS", which would allow up to 999 minutes (16 hours) to be displayed without sacrificing the seconds precision.

Arvycka, 2020-08-14:
Or jus get rid of the 2 g-force lines?

breadoven, 2020-08-14:
And for those who regularly forget to press record before arming having Time and Date on the DISARM screen would be useful.

OptimusTi, 2020-08-14:
How about adding a second page and grouping data by topic (Power, Navigation, Etc.) 2 seconds each page?

Jetrell, 2020-08-15:
> 
> 
> Or just get rid of the 2 g-force lines?

The G-force stats are nice to know. But last I looked, Maximum Z & X axis G-Force was recorded a touch down. So that makes the G-force Stats not as useful as their log data.
So I would also agree with the removal of G-force, unless a filter has been added to fix the landing issue.

danarrib, 2020-08-24:
> This was my flight, it disarmed in air and im pretty sure i didnt flicked the disarm switch. It stayed in statistics screen for about 1 minute and didnt let me arm the model. I checked the log and i saw alot of momentary disarms and arms, it disarms for about 20 millisecond and then arms again, i think it also initiates rth too. I still dont know why this happens and how to solve it. Here is the logs if anyone interested, log24 before disarm, log25 after arm.
> https://mega.nz/file/DCJSQYgA#-S5m2tedPuVjafnuxP7_jvamrnengyH1Q0DcyVtq10w
> https://mega.nz/file/vLYmQCZR#U6E6A2c9t2xXA4iczXTU6fziLGOLDoOUnFKesT_BYZs

Welcome @jetstream123 .
I saw your screenshot on Facebook and based on that I spotted the bug on the OSD screen. This issue will address the bug on the OSD screen and not the disarm problem itself.

About your disarm problem, it looks like a bad contact on your transmitter switch. The log file tells that the channel value in fact is switching, which indicates a problem on the radio transmitter itself.

INAV does have a feature to prevent accidental disarms due channel value glitches, it's "`switch_disarm_delay`", the default value is 250, which gives 1/4 of second of tolerance. You should check if your transmitter switch is working fine and replace the bad switch if it's the case.

danarrib, 2020-09-03:
Another solution is to allow user to suppress the lines that he thinks that are not so useful. I guess it's easier to do.

avsaase, 2021-02-01:
I am trying to make some improvements to the disarm screen in #6548. Feedback is welcome. I think betaflight has fully configurable disarm screen, but implementing that is a bit beyond my current skills.

OptimusTi, 2021-02-01:
I've been thinking of doing the same. Something like what KISS has.

OptimusTi, 2021-02-09:
[image]

This should help with the issue and adds extra lines for other stuff (ESC Temp, RPM?). You can change pages with the sticks. No need to remove lines or cram stuff

danarrib, 2021-02-09:
That's super nice!

OptimusTi, 2021-02-09:
I'm following #6548 to see if there's any progress. If not, I'll send a PR soon.

avsaase, 2021-02-09:
@OptimusTi This is a much better solution so please go ahead and create a PR for this. I still would like to propose that max current/power is one line, but current voltage should be a separate line.

OptimusTi, 2021-02-10:
PR coming in a few minutes. Cheers

avsaase, 2021-02-10:
As commented by @Jetrell on my (now closed) PR:

> Another tidy up suggestion. Instead of the Amps, Watts and Power unit letters being alphabet characters.
Can you replace them with font characters
>
> A with 0x9a
> V with 0x06
> W with 0xae
> to keep more inline with the rest of the stats?

stuwe911, 2021-09-22:
I really can't get warm with that 2-pager stats osd besides that i can't switch back to page 1/2 once i've switched over to page 2. For me this is a reason to downgrade back to 2.6.1. Why not just add a flag at each available attribute so that uses can "customize" their stats themselves instead of all these discussions of what should be shown and what is not needed? that would solve everybody's issues once and for all.

OptimusTi, 2021-09-22:
This was merged months ago, and feedback has been good.

1. Roll left or right to change pages
2. From 3.0.1, pages auto roll

I've been thinking of changing to something like what BF is using. Just click the stats that you need. This way we could only add pages if needed. Maybe even allow the user to sort stats in the configurator by dragging things around. But time is limited and my Javascript is limited.

OptimusTi, 2021-09-22:
@danarrib close or change to a discussion?

danarrib, 2021-09-22:
I am not who decides how the Stats page WILL be... I can, at most, make suggestions. The 3.0 status page looks fine for me as it is... It pages all stats automatically and allow user to change pages if he/she wants. 

I should have closed this issue on February when #6598 was merged, but somehow I missed that. If @stuwe911 or anyone else wants to suggest another approach for the OSD stats, please raise a new issue.

closed 2021-09-22 by danarrib

stuwe911, 2021-09-22:
to me it looks perfect what Optimus is proposing. i've tried all the sticks left/right whatever. once i'm on page 2 it will stay there until i arming again. (using 3.0.2). But I can live with downgrading back to 2.6.1. that does the trick for me.

OptimusTi, 2021-09-22:
> to me it looks perfect what Optimus is proposing. i've tried all the sticks left/right whatever. once i'm on page 2 it will stay there until i arming again. (using 3.0.2). But I can live with downgrading back to 2.6.1. that does the trick for me.

Rolling left will not go back to page 1? Have you limited your endpoints somehow?

stuwe911, 2021-09-22:
2 times a no. even tried the rudder and switches. changing from page 1 to 2 works fine but then it stays there until reboot or arming again.. maybe it's something with my setup as it seems to work for anyone else. but I really don't see the point of 2 half pages when it was all just fine before. even for the NTSC guys. getting more stuff onto the stats could make sense but that is so individual like flying. If you're a aerobatic guy you're interested in the g's. normal pilots could not care less. so there is no perfect stats page that fits everyone. hence your proposed solution would kill many birds with one stone rather than just a few developers deciding for the whole world.
I keep a logbook for all my long range flight so the stats are kinda important to me. As it turns out more than the new features and improvements v3 offers. So i will go back to 2.6.1 and upgrade when your solution of a customized stats page is implemented or my hardware support runs out. no biggie. but 2x half a page just does not make sense.. at least to me. but i guess that is just a personal preference and not the end of the world.
thanks for your great proposal!

OptimusTi, 2021-09-22:
Must be something with your setup. To exit you can pitch "high" or throttle "high" 

As for adding more lines, yes, there were a couple of lines added for the crsf protocol.

Here's an updated gif but someone else needs to work on it. Lots of changes needed.
[image]
.

stuwe911, 2021-09-22:
that looks just soo great if that was possible what's shown in your gif. i hope someone will work on it. i was wondering why i still have RSSI in the stats even though i've changed over to LQ from CRSF... no biggie but it would be a very very nice to have.
thanks