iNavFlight/inav #6048: Setting the time zone to adjust GPS date time

Pairan, 2020-08-14:
## Current Behavior
Currently the date and time information are delivered by the GPS and handed over "as-is" to telemetry. The OSD shows the time from GPS. This may differ from the time zones the pilot my be in. 

As I don't know if this is already possible (used goggle but didn't find something) I decided to place this feature request.

## Desired Behavior
By adjusting this information the correct time could be displayed in the OSD

## Suggested Solution
It would be a nice features to be able to adjust the date time values from the GPS according to the time zone - perhaps the GPS is already delivering these data (e.g. like central Europe would be +2h). Otherwise a CLI setting or configurater field could do the job

## Who does this impact? Who is this for?
Pilots who want to see the correct(ed) GPS time in their OSD

## Additional context
This is suggest to be a FEATURE

OptimusTi, 2020-08-14:
Look up tz_offset and tz_automatic_dst. 

tz_offset = Time zone offset from UTC, in minutes. This is applied to the GPS time for logging and time-stamping of Blackbox logs. (Default = 0)

tz_automatic_dst = Automatically add Daylight Saving Time to the GPS time when needed or simply ignore it. Includes presets for EU and the USA - if you live outside these areas it is suggested to manage DST manually via `tz_offset`. (Default = Off)

If you ask me, this should be a inav-configurator request.  :)

Jetrell, 2020-08-15:
> If you ask me, this should be a inav-configurator request. :) Perhaps adding "UTC Offset" [-12..12] to the GPS area in the Configuration tab?

@OptimusTi  I think what @Pairan was getting at... Was adding a table. That automatically calculates the UTC offset according to the GPS Longitudinal coordinates.

Pairan, 2020-08-17:
I didnt know of `tz_offset` and `tz_automatic_dst` - thanks a lot! 

so for me it's
`
set tz_offset = 2
set tz_automatic_dst = ON
save
`

And yes- a widget inside the configurator tab "CONFIG" would be nice ...

closed 2020-08-17 by Pairan

OptimusTi, 2020-08-17:
> I didnt know of `tz_offset` and `tz_automatic_dst` - thanks a lot!
> 
> so for me it's
> 
> `set tz_offset = 2`
> `set tz_automatic_dst = ON`
> 
> And yes- a widget inside the configurator tab "CONFIG" would be nice ...

Make sure you read the documentation.  tz_offset is in minutes.

Pairan, 2020-08-18:
thanks for the heads up - i missed that. then it must be 120 (2h)