iNavFlight/inav #6051: Create mid-point TPA (negative or reversed TPA)

stuwe911, 2020-08-15:
## Current Behavior
Since i'll do lots of glided dives with planes I'm struggling getting my PIFF settings right. I always have to find the "middle way" which never works great in either situation. In order to prevent oscillation in a dive i have to set the wonderful feature of stabilized flying so low, that in normal cruise I can hardly feel it. I even had short approach emergency landings where the plan started to oscillate 10m above the ground - was scary! I'm not the type that most of the time flies with 80-90% throttle so that TPA would really give me a great benefit - i need it the other way around. Kinda like a "negative TPA" or even better a "mid-point TPA"

## Desired Behavior
Create a mid-point TPA that works proportionally before and after reaching the set break point. i.e. i set "mid-point" TPA to 1500 with a value of 50%. So at 0 throttle i would have 50% PIFF/PID effect, 10 throttle, 60%, 20 throttle 70% etc. until i reach the set 50 throttle (1500pwm) where the full PIFF/PID settings work. From there upwards it would work just like the regular TPA today like 60 throttle 40%, 70 throttle 30% etc.

## Suggested Solution
Introduce an option to select "standard TPA" or "mid-point TPA"

## Who does this impact? Who is this for?
Anyone who struggles with oscillation when gliding and paying the price with not enough "locked-in" when flying normally with 40-60% throttle

## Additional context
It might be simpler (faster implemented) by just adding a "reversed" or "negative" TPA so either or / both can be activated?

DzikuVx, 2020-08-16:
This is how INAV TPA works on fixed wings. Please read https://github.com/iNavFlight/inav/wiki/PID-Attenuation-and-scaling

closed 2020-08-16 by DzikuVx

stuwe911, 2020-08-16:
Pawel, thank you for that link. i will play around more with the TPA settings according to that document in order to get the desired result. However this kinda contradicts with what i've read on the INAV Fixed Wing Group Master Class Tuning page where they also state: "How INAV implements the TPA right now is not perfect, and can’t be optimised for all three situations. You need to find the correct balance between fast flight, slow flight and glide dives." This is exactly how i've experienced that setting.. either - or - or a little bit of everything but nothing real. However I will keep up the testing go get at lease the slow flight and glide dives right.
cheers

stuwe911, 2020-10-19:
Pawel,
in the meantime i have solved my TPA problem by deactivating it. I would
still need the "both ends scaling" solution in order to use it due to high
speed 0 power dives where it's still too much currently. here you can see
why / where i need that. maybe you like it
cheers
Stuwe

https://youtu.be/VFNZreeDc_c