iNavFlight/inav #6052: Smartport Telemetry Not Updating on Taranis

Davierooker, 2020-08-16:
**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior
Sensor discovery slow and sensors not updating on Taranis.

## Steps to Reproduce
1.Flash Matek F411 Wing and F405 Wing to 2.5.2
2.Setup Smartport on softserial
3.Discover sensors on Taranis with R9
4.Sensors are discovered randomly or not at all, any that are dicovered don't update leading to "sensor lost" alerts.

## Expected behavior
Sensors should be discovered instantly, and have a heartbeat.

## Suggested solution(s)

## Additional context
Flashed back to 2.2 and discovery works

---

2.5.2

- FC Board name and vendor:Matek f411
- INAV version string:INAV/MATEKF411 2.5.2 Aug  4 2020 / 10:25:53 (faaedc7cb)

Davierooker, 2020-08-17:
After further investigation the fault appears to be related to SmartPort when used over SoftSerial on F4. I have changed my configuration to FBUS and the issue has been resolved. The issue does not seem to affect F7 FC's when using SoftSerial.

janmj, 2020-09-02:
I noticed this issue too lately. Began noticing after 2.3 i had a lot more "sensor lost" warnings than normal, and in some cases smartport telemetry works for a few second but then stops completely until a reboot. I have a lot of Omnibus F4 pro's i use on smaller planes and wings, and getting f.port on these is a pain because of inversion. But worse is that since 2.4 it have started happening on my Matek 405 wing's too. Takes longer, but after 15-20 min's i get "sensor lost", and no more telemetry until i reboot. Makes using my U360GTS tracker all but imposible :(.

Davierooker, 2020-09-08:
Are we getting to the limit of F4 processors by asking to handle SoftSerial on newer builds of INAV?

janmj, 2020-09-08:
Perhaps we are. But processor load do not indicate any strain. And was there really som much new functionallity from 2.2.x to 2.3 that it would overload a F4 so much that softserial is all but unusable for telemetry?

janmj, 2020-09-08:
Just to be clear. This is only for a wing or a plane. I have yet to try INAV on a quad.

j1r1, 2020-09-09:
The same problem met with Wing FC without any SoftSerial port used, any XJT or R9 receiver and Jumper T16. I had to return to v2.3.

Quadropus, 2020-09-10:
Same problem with Matek F765 and FrSky RX6R with SmartPort on versions 2.5.2, 2.5.1 and 2.5.0. 
SmartPort telemetry works as intended on version 2.4.0.

j1r1, 2020-09-11:
My problem solved by telemetry_halfduplex =OFF.

janmj, 2020-09-11:
I suppose you mean "telemetry_uart_unidir"? "telemetry_halfduplex" isn't in INAV. Anyway i have both "telemetry_uart_unidir=off" and "serialrx_halfduplex=off". Always have on F4 boards.

j1r1, 2020-09-11:
That may be the core problem. In my 2.5.2 firmware for WingFC (and for OMNIBUS F4 PRO too) definitely IS "telemetry _halfduplex" and is NOT "telemetry_uart_unidir".
I am not sure whether it helped, but before installation of the new version of configurator I deleted Windows directory of \Users\...\AppData\Local\inav-configurator.

janmj, 2020-09-11:
That's correct. I forgot to give my version. 2.4. Haven't bothered to upgrade to 2.5.x yet. But. it's just a name change, not different functionality: https://github.com/iNavFlight/inav/pull/5455

Quadropus, 2020-09-11:
Telemetry_halfduplex is turned off in my case. Like recommended in the Telemetry Wiki section. (there refered as "telemetry_uart_unidir = OFF")
I made a completely new install for the iNav Configurator and the problem occured immediately.
I noticed telemetry sensor loss in V2.4 and V2.3 as well after some time.

IdefixRC, 2020-09-13:
I can confirm the same issue with an OmnibusF4 Pro and the Wing FC-10. 
No issues on 2.4.0, as soon as I go to 2.5.x smart port telemetry stops working

j1r1, 2020-09-14:
Although it worked for me for some time, now I am not able to get the telemetry running on WingFC+R9 Slim OTA+Jumper T16 with INAV v.2.5.2. I confirm the issue. After returning to v.2.3.0 works fine.

ehubin, 2021-01-28:
I have a similar problem on a Matek F411 WSE running INAV 2.6.1 where a R9 sx is connected to the Tx2 pin configured as softserial. No FC sensors are detected on Taranis Transmitter apart from the Receiver ones.
After a bit of flashing, it seems the problem was introduced in 2,5.0 as sensor detection is instantaneous in 2.4 anddoes not work at all in 2.5.0.
I have looked into the commit list and it seems there has been rework on FrSky Crc algorithm between these 2 versions.
Any idea whether this could be related?

sergiobd007, 2021-02-06:
I have the exact same issue as OP, it's working in 2.3 but not working in any later version, I'm using a Matek F411 WING as well.

I used the workaround of updating the firmware in my receiver with FPort and connecting the FC with the inverted signal from the receiver (had to solder in a specific point on the receiver to get the uninverted signal, you can look up the specifics online). I'm now using the latest version of inav without problem, but it's a pain.

stsa64, 2021-03-05:
I have the same issue on 2.5.2, Matek F405 Wing, Smartport on SoftSerial, R-XSR receiver, Taranis X9D. No sensors detected beyond receiver. Would be great if someone could investigate. Happy to help if needed on e.g. trying out a bug fix. I'm not too keen on downgrading in order to fix this.

stsa64, 2021-03-15:
@sergiobd007 : What INAV version are you running with Fport where everything works?
I switched to Fport (what a hassle to solder...) on 2.5.2. and it seemed to work (at least when doing a quick test). Then I upgraded to latest (2.6.1.) and now the sensors are detected after some time, but they are lost quickly/are not really updating.

stsa64, 2021-03-16:
I did some further testing. When I reboot the FC, the sensors (beyond the 2 receiver sensors) are discovered/provide data normally after RC link is established for about 2 seconds. Around the time the startup sequence (beeps) is finished, the sensors no longer get updated ("sensors lost" warning). So to me, it seems everything is initialized correctly but then goes awry in "run" mode. I tried this many times now, consistent result every time. Ideas anyone_

Current config: INAV 2.6.1, Matek F411 Wing, FrSky R-XSR running Fport on UART 2 via uninverted smartport output, FrSky X-lite Pro.

janmj, 2021-03-16:
I have seen exactly the same behaviour. Fresh reboot ok, but after a few seconds it stops updating. I noticed this does not happen when using f-port on the same reeiver, so i'm switching over all my receivers wich support f-port.

stsa64, 2021-03-16:
@shellixyz : Hi Michel, the Smartport issues here may have been introduced as part of the work on the Smartport CRC algorithm last year (#5641). Would be great to get your view on this! Thanks.

stsa64, 2021-03-17:
Spent some more time investigating: I can get things to work (still Fport) on all versions from 2.3 to 2.6.1. with clean installs. HOWEVER, it stops working as soon as I enable the GPS (at least on 2.6.1). I guess this indicates the telemetry chokes on the GPS data. I tried to make a build with the previous CRC code, this seems to make no difference, so currently, I do not believe the error is there.
**Previous posters**: do you have GPS? Can you try to disable it and see if it makes a difference? Would be good to understand if this is one issue or two separate ones.

sergiobd007, 2021-03-17:
It works fine for me when using Fport with an XSR receiver on 2.6.1. I'm using GPS without any issue as well.

stsa64, 2021-03-19:
I hacked away at the code and combining this with further investigation online my current issue with Fport / sensors lost seems to be unrelated to the OP issue. 

Just FYI in case this issue goes in that direction: it seems Smartport can get problems when there are too many sensors. Notably there was an issue with Frsky R-XSRs (which I use now) which was discussed in Betaflight forum (should have been fixed via RX firmware). Right now I can get everything to work if I disable some of the sensors (doesn't seem to matter which ones) in the Smartport code...

Viper1200, 2021-04-22:
Same telemetry issue for me using Jumper T16 & Radiomaster 168 (or a 161) with a FC F411 WSE. Connections are SBUS & Smartport.
I have the GPS enabled but get the same problem with it disabled as well.

KickassFPV, 2021-04-30:
set telemetry_halfduplex = ON worked for me

Tommy032017, 2021-05-18:
> @stsa64 :  Spent some more time investigating: I can get things to work (still Fport) on all versions from 2.3 to 2.6.1. with clean installs. HOWEVER, it stops working as soon as I enable the GPS (at least on 2.6.1). I guess this indicates the telemetry chokes on the GPS data. I tried to make a build with the previous CRC code, this seems to make no difference, so currently, I do not believe the error is there.

Hi Steffen, I've bought 3 F411 WING and also 3 R-XRS and tested everything to get it work with Fport (soldered to P), but I have the same problem, that telemetry will stop, as soon as I enable GPS. I investigated a lot of time to find a solution, tried Betaflight, Inav, the latest INAV 3.0 and also firmware FRSKY ACCST D16 v2.1.1 and 2.1.0, changed from UART1 to 2 and vice versa, but nothing could solve the problem. I also tried all the F411 and R-XRS, but the same problem exists to all my boards. It only works, if GPS (doesn't matter to serial port settings and/or GPS protocol) is enabled.

Do you found a solution to this, which works fine with all sensors? 
You found a solution by disabling some sensors. Could you pls. tell me how to disable them or may be provide me your firmware with disabled sensors.
Thanks

stsa64, 2021-06-10:
> > @stsa64 :  Spent some more time investigating: I can get things to work (still Fport) on all versions from 2.3 to 2.6.1. with clean installs. HOWEVER, it stops working as soon as I enable the GPS (at least on 2.6.1). I guess this indicates the telemetry chokes on the GPS data. I tried to make a build with the previous CRC code, this seems to make no difference, so currently, I do not believe the error is there.
> 
> Hi Steffen, I've bought 3 F411 WING and also 3 R-XRS and tested everything to get it work with Fport (soldered to P), but I have the same problem, that telemetry will stop, as soon as I enable GPS. I investigated a lot of time to find a solution, tried Betaflight, Inav, the latest INAV 3.0 and also firmware FRSKY ACCST D16 v2.1.1 and 2.1.0, changed from UART1 to 2 and vice versa, but nothing could solve the problem. I also tried all the F411 and R-XRS, but the same problem exists to all my boards. It only works, if GPS (doesn't matter to serial port settings and/or GPS protocol) is disabled.
> 
> Do you found a solution to this, which works fine with all sensors?
> You found a solution by disabling some sensors. Could you pls. tell me how to disable them or may be provide me your firmware with disabled sensors.
> Thanks

Hi Tommy,
Sorry for the late reply. Unfortunately my solution only worked partly in the end. When I altered the firmware, it seemed to work for some time, but after flying, it still stops working after some minutes. My hypothesis is, that our issue is linked to the R-XSR receivers. Unfortunately I do not have other receiver models to verify.

closed 2022-02-23 by DzikuVx