iNavFlight/inav #6053: Zero throttle in CRUISE when nav_fw_allow_manual_thr_increase = ON and throttle stick at zero

avs2, 2020-08-16:
# Current Behavior
In `CRUISE` mode throttle in managed by INAV using `nav_fw_cruise_thr` and `nav_fw_pitch2thr`. When `nav_fw_allow_manual_thr_increase = ON` you can manually further increase the throttler by raising the throttle stick above the current auto throttle level. If the trottle stick is lower you get the auto throttle level (at least, that is how I understood it when I recently asked in the rcgroups thread). However, when you set the throttle stick to zero this seems to override the auto throttle level and causes zero throttle output. This is also visible in the OSD. If you don't pay attention this can easily stall and crash your airplane.

The problem does not occur in `RTH`. I have not tested if this also occurs when `nav_fw_allow_manual_thr_increase = OFF`.

## Steps to Reproduce
1. Enable `nav_fw_allow_manual_thr_increase = ON`. 
2. Fly in `CRUISE` mode.
3. Set throttle stick to zero
4. See that throttle is reduced to zero instead of the auto throttle level.

## Expected behavior
When throttle stick is between auto throttle level and zero (inclusive), throttle output should be the auto throttle level.

## Additional context
[diff](https://github.com/iNavFlight/inav/files/5081381/INAV_cli_Talon_GT_20200816_224252.txt), [blackbox log](https://github.com/iNavFlight/inav/files/5081390/LOG00023.TXT) and [dvr footage](https://youtu.be/zSOit6MihTc).

---

- FC Board name and vendor: Matek F405-Wing
- INAV version string: INAV/MATEKF405SE 2.5.2 Aug  4 2020 / 10:24:57 (faaedc7cb)
GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

breadoven, 2020-08-16:
I think only RTH and WP Nav modes keep the motor running under Auto throttle when the throttle stick is below Min Check, otherwise the motor stops. nav_fw_allow_manual_thr_increase ON or OFF makes no difference. However I discovered recently that the new Global Function throttle override will kill the motor even in RTH mode which you need to be aware of if you want to avoid surprises.

jseaber, 2020-08-17:
I encountered the same bug today during a test flight on INAV 2.5.2 and a Matek F405-WING, resulting in an unintended landing. `nav_fw_allow_manual_thr_increase` is not utilized in my setup.

Here's `NAV ALTHOLD` + `NAV CRUISE` correctly functioning early in the flight, above an AMA airfield. Note the Auto Throttle value in bottom left corner of the OSD is working:
[image]

RTH operated successfully with radio powered down later in the flight. Auto Throttle value is correct in this mode as well:
[image]

In order to power up my radio, I must set throttle to 0. When I took over control and set to 3CRS again, Auto Throttle remained pegged at 0:
[image]

Unfortunately, I did not catch this bug as I had lifted my goggles to check the runway. Here's the wing safely landing in a bean field while I trusted it would hold altitude:
[image]

No harm done, but this is a serious bug!

sakisdog, 2020-08-18:
#5390

avs2, 2020-08-18:
> #5390

I searched for similar (open) issues before opening this one but since #5390 is closed I missed it. Anyway, I think this is a pretty serious bug so I'll leave this issue open so the devs can take a look at it.

jdslabs, 2020-08-18:
@avs2 - Totally agree. If this is a misunderstanding of intended functionality, it certainly deserves more attention on the OSD. A flashing Warning message would be adequate.

jseaber, 2020-08-18:
@avs2 - Totally agree. If this is a misunderstanding of intended functionality, it certainly deserves more attention on the OSD. A flashing Warning message would be adequate.

sakisdog, 2020-08-18:
You could try to set min_check to 1000 so that the code wouldn't return THROTTLE_LOW when the throttle stick is at zero position. Else you should modify the code.

jseaber, 2020-08-18:
@sakisdog I am an engineer, enjoy iNav, and take the time to read setup documentation. I've been flying for a couple years, with hundreds of flights under my belt and a part 107 license, etc.

Even with several spotters at the field, I had placed trust that Altitude Hold would do its job. Because of this programming choice, the wing landed and nearly clipped an electrical line on its way down.

Yes, I can think of a myriad of solutions here. Ultimately, this  this is a safety concern that can either be addressed clearly in iNav documentation, or through OSD Warning.

I'll be glad to dive into the code. @avs2 Any feedback?

avs2, 2020-08-18:
I don't see any logical reason why the current behavior would be desirable, so I assume it's just a bug. Only the devs can confirm this but I understand it is vacation time so we will just have to wait.

From #5390 I understand that the issue is independent from the `nav_fw_allow_manual_thr_increase` setting but I haven't tested that myself. The documentation is a little unclear on `nav_overrides_motor_stop` because the motor_stop parameter was removed a couple versions ago but to me it seems that the motor should not be stopped in the nav modes, including `CRUISE`.

sakisdog, 2020-08-19:
@avs I spent 3 days reading the code to understand why the motor would not spin while i was trying to implement a feature of my own.

I don't know if this is a bug or an intended behavior as i have zero experience with airplanes, fixed wings etc.

What i know is that the motor stops because of the following code:
`if (calculateThrottleStatus(feature(FEATURE_REVERSIBLE_MOTORS) ? THROTTLE_STATUS_TYPE_COMMAND : THROTTLE_STATUS_TYPE_RC) == THROTTLE_LOW) {
        if ((STATE(FIXED_WING_LEGACY) || !STATE(AIRMODE_ACTIVE)) && (!(navigationIsFlyingAutonomousMode() && navConfig()->general.flags.auto_overrides_motor_stop)) && (!failsafeIsActive())) {
            DEBUG_SET(DEBUG_POSHOLD,0,1);
            return MOTOR_STOPPED_USER;
        }
    }`

This piece of code prevents the auto throttle to work when the throttle stick is at 0 position (THROTTLE_LOW) because navigationIsFlyingAutonomousMode function is true only with RTH and WP missions so when throttle is lower than min_check(default is 1100) and you are not flying autonomous (RTH or WP) then it returns MOTOR_STOPPED_USER.

breadoven, 2020-08-19:
If you look in **navigation.c** around line 3500 **"bool navigationIsFlyingAutonomousMode(void)"** (also note the routine above which is also relevant):

```
bool navigationIsControllingThrottle(void)
{
    navigationFSMStateFlags_t stateFlags = navGetCurrentStateFlags();
    return ((stateFlags & (NAV_CTL_ALT | NAV_CTL_EMERG | NAV_CTL_LAUNCH | NAV_CTL_LAND)) && (getMotorStatus() != MOTOR_STOPPED_USER));
}

**bool navigationIsFlyingAutonomousMode(void)
{
    navigationFSMStateFlags_t stateFlags = navGetCurrentStateFlags();
    return (stateFlags & (NAV_AUTO_RTH | NAV_AUTO_WP));**
```

Only RTH and WP are defined as Autonomous which prevents MOTOR-STOPPED_USER happening in **Mixer.c**:

```
motorStatus_e getMotorStatus(void)
{
    if (failsafeRequiresMotorStop() || (!failsafeIsActive() && STATE(NAV_MOTOR_STOP_OR_IDLE))) {
        return MOTOR_STOPPED_AUTO;
    }

    if (calculateThrottleStatus(feature(FEATURE_REVERSIBLE_MOTORS) ? THROTTLE_STATUS_TYPE_COMMAND : THROTTLE_STATUS_TYPE_RC) == THROTTLE_LOW) {
        if ((STATE(FIXED_WING_LEGACY) || !STATE(AIRMODE_ACTIVE)) && (**!(navigationIsFlyingAutonomousMode**() && navConfig()->general.flags.auto_overrides_motor_stop)) && (!failsafeIsActive())) {
            return MOTOR_STOPPED_USER;
        }
    }

    return MOTOR_RUNNING;
```

So this appears to be by design not a bug and I seem to remember discussion about it before where motor safety was considered the most important thing. To me it's confusing and illogical as it is. To have Nav modes take full control of throttle except when you go to Low throttle makes no sense. If you really want to stop the motor urgently then Disarm or switch off the Nav mode. You can't enable these modes on the ground anyway (or can you, maybe I never tried ?) so it's not as if the motor is going to suddenly start up unexpectedly close to people. Or is this confused by differing requirements for Fixed Wing and Multirotors ? What does a Multirotor do in Althold/Poshold with throttle low ? Drop ?

sakisdog, 2020-08-19:
@breadoven This happens only with FIXED_WING_LEGACY (AIRPLANE, ROVER, BOAT) and not with multirotors. The function checks if STATE(FIXED_WING_LEGACY) is true AND if navigationIsFlyingAutonomousMode is false. Then it returns MOTOR_STOP. So in the case of Multirotors POSHOLD controls the throttle even when the stick is at zero position.

breadoven, 2020-08-19:
Yes, should have seen that bit regarding FIXED_WING_LEGACY although given it's obvious a MR needs to keep the motors running for it to stay in the air it begs the question ... what's the difference between allowing motors on a MR to continue running in all Nav modes regardless of throttle position but not on a Fixed Wing ?

sakisdog, 2020-08-19:
> what's the difference between allowing motors on a MR to continue running in all Nav modes regardless of throttle position but not on a Fixed Wing ?
That's indeed a very interesting question that needs to be answered by some exprerts.

shellixyz, 2020-08-24:
As @breadoven said it is not a bug. This is intended. I agree that a setting to turn that on which defaults to off would be a good idea though

closed 2020-08-24 by shellixyz

avs2, 2020-08-24:
I fail to see how this is not a bug. I have `nav_overrides_motor_stop = ON` (the default). Does that not mean that the navigation modes should override motor stop? Currently it clearly doesn't.

shellixyz, 2020-08-24:
> `nav_overrides_motor_stop`

This setting has an effect only on autonomous modes aka `NAV RTH` or `NAV WP`

avs2, 2020-08-24:
I would argue that `POS HOLD` and `CRUISE` should follow this setting to be consistent since they are (partially) autonomous modes as well. I don't object to have more options but the current behavior is what caught me and @jseaber by surprise.

shellixyz, 2020-08-24:
We could indeed just extend `nav_overrides_motor_stop` to work for all nav modes controlling the motor in my opinion. I asked the developer who added this setting why he made it only work for `RTH` and `WP`

digitalentity, 2020-08-24:
Currently `nav_overrides_motor_stop` only affects fully autonomous modes for consistency - to give the pilot the same zero throttle behavior when the pilot is in control of the throttle stick.

We discussed this and we'll be changing this to a 3-state with options to ignore motor stop for all navigation modes (regardless of pilot-controlled or automated); only for automatic navigation modes; and disabled. This should cover all scenarios.

avs2, 2020-08-24:
That would be perfect. Thanks!