iNavFlight/inav #6057: Target request: SucceX-D F7 V2.1 TwinG Flight Controller

bckenneth, 2020-08-19:
Is that possible create a target for SucceX-D F7 V2.1 TwinG Flight Controller for Inav.
Thanks

Lusby, 2020-11-25:
+1
the quad this FC is in, Chimera7, is a perfect candidate for iNav (may need a GPS upgrade)

schmidi-dev, 2020-12-31:
Whats the status? I'm confused by contradictory information.

vpominchuk, 2021-01-30:
+1 
very interesting and affordable drone made by iFlight on this controller. It's will be great if we can make it much better flashing INAV. Thank you.

MaikFPV, 2021-03-22:
+1
please add target - stay healthy

prohelper, 2021-03-27:
> 
> 
> is this the target? **inav_2.6.0_IFLIGHTF7_TWING**

To what i can see this target doesnt have UART 3 for some reason. Is there anyway mabye they can update the target?

janmj, 2021-04-29:
> > is this the target? **inav_2.6.0_IFLIGHTF7_TWING**
> 
> To what i can see this target doesnt have UART 3 for some reason. Is there anyway mabye they can update the target?

Uart 3 is used for i2c. This is mentioned on the wiring schematic on the iflight site. But is the current inav target otherwise functioning on the V 2 1 board?

prohelper, 2021-05-27:
I dont have access to my drone at the moment. I remember the ports being different from betaflights config

mtshare, 2021-05-27:
@prohelper so the target is correct (**IFLIGHTF7_TWING**)?

nowaythatsok, 2021-09-15:
Does anyone have an update on this? Is there a target now for this FC?

closed 2022-02-10 by DzikuVx

MaikFPV, 2022-03-07:
Closed = implemented? Can someone confirm working target? Thanks a lot!

ridgeracer, 2022-07-28:
I use this target for an older Iflight F722 TwinG BT. Can’t confirm if compass will work on Uart3 which should be SDA/SDL. GPS doesn’t work on Uart4 or Uart5. Conflicts with M1 servo output. Currently set M3 to dshot150. GPS works on Uart2. RX works on Uart1. Servo outputs confirmed working on M1 - M4. So this target can use full Uarts on 4 or 5. Can others confirm same?