iNavFlight/inav #6065: Make gliding at 0 throttle in althold modes an option

shellixyz, 2020-08-24:
When the throttle is bellow `mincheck` in any nav mode doing altitude control (nav states using NAV_CTL_ALT) the motor is shut off and the model allowed to glide down. This behavior can catch users by surprise and should be made an option, turned off by default.

Issue where this was brought up: #6053

breadoven, 2020-08-24:
Just to complicate things, how does this fit in with the Global Functions that override throttle control, Throttle Scaling and Throttle Override. For motor glider users It would be useful to switch off Alt_Hold for all instances when the throttle is stopped, using Global Functions or below Min_Check,  if this new option is added.

I also discovered that using Throttle Scaling to stop the motor above Min_Check causes the motor to remain stopped when you select RTH (switched not failsafe). Easily fixed with an extra logic condition but is this desirable when using these throttle GFs ?

breadoven, 2020-08-24:
I have flaps mixed into the throttle stick (no sliders on my transmitter) when the flaps switch is ON so the flaps can be controlled by the throttle stick,. In order to use the flaps when soaring/thermalling you want to use the throttle stick without the motor running so I have is another "soaring" switch that triggers throttle scaling to stop the motor/scale power above Min_check. If iNAV reads this scaled throttle as if it's coming from the stick itself then it should all work out as required, i.e. if the stick is MID (motor stopped by throttle scaling of 0 in my case) and iNAV reads throttle as 1000 then all is OK, but if it reads throttle as 1500 then NAV_CTL_ALT won't be disabled and you can't glide down (or soar up in fact).

Perhaps I should just buy a better transmitter although using the throttle stick to control flaps when soaring is better then a slider IMO.

Not sure about the RTH issue though because it shouldn't stop the motor even below Min_check should it, or is that only true for Failsafe RTH not switched RTH ?

shellixyz, 2020-08-24:
Isn't it possible with your transmitter to force the throttle to 0 when you flick your soaring switch ? Seems like the obvious solution to me if possible. If you have a transmitter running OpenTX it is possible to do that. Of course if it is not possible for you it makes things more complicated.

Hmmm I never used the global functions to do such thing so I just took a look at the code again to see what would really happen if you tried to just set the throttle through them. Indeed it looks like it is not going to work because getMotorStatus() is using calculateThrottleStatus() to get the throttle state and that's not going to be influenced by the GF override. Perhaps there is something to do there. GFs are relatively new.

About failsafe RTH it will always try to use the motor. Though if you altered the scaling with GFs failsafing probably doesn't change it back to 1/1 so the command is always bellow min_check and the motor never spins. Seems to me that using scaling to do what you're trying to do is quiet dangerous. In the case of FS RTH INAV will consider it has thrust, will try to keep the target altitude and it is quiet basic. If the plane goes under the target more and more it will try to pitch up more and more without being able to get thrust and the plane will most likely end up in a stall.

The solution I think is to allow the global function throttle override to stop the motor.

breadoven, 2020-08-24:
Flysky i6 Tx. Don't think you can override the stick outputs.

I was just wondering if it's possible to take advantage of NAV_CTL_ALT being disabled with low throttle which I didn't realise even existed until now. But no big deal if it complicates things.

I use Throttle Scaling of 0 up to 75% stick then 50% scaling above 75% so there is some power if you need it. I wasn't aware Throttle Override was available when I set it up originally so used scaling of 0 instead. I could in fact use Override so may well change it. However, I added a Logic Condition making the Throttle Scaling dependent on RTH not being selected so RTH now works as required.

PR #6067 (2020-11-14): Make nav_overrides_motor_stop tristate: OFF | AUTO_ONLY | ALL_NAV
When throttle is low:
* OFF = no override
* AUTO_ONLY = override only in autonomous modes: RTH and WP
* ALL_NAV = override in all nav modes

Closes #6065
closed 2020-11-14 by digitalentity