iNavFlight/inav #6066: SBUS Output + PWM Output

IVData, 2020-08-24:
## Current Behavior
In 2.5 you can choose either SBUS output **or** PWM output

## Desired Behavior
Allow both SBUS output on a UART and PWM output from the normal pins.

## Suggested Solution
Since the sbus output is a serial function can the PWM output be left working?

## Who does this impact? Who is this for?
My use case is e.g. a arduino-controlled stepper motor pan with a servo tilt. The arduino can receive sbus and use 3 channels for pan, tilt, and a centering switch without having to use 3 separate PWM outputs.

IVData, 2020-08-24:
I can imagine it'll be quite common that when someone enables sbus output for use with sbus servos, they may want to also have PWM output for a normal servo, nav lights, PWM switches etc. and it would be great to be able to use the existing pins on the FC for this.

DzikuVx, 2021-03-02:
#6106

closed 2021-03-02 by DzikuVx