iNavFlight/inav #6069: No Time on GPX file - Cannot view flight in Google Earth

AdamGFPV, 2020-08-25:
## **Current Behavior:** Blackbox logging to built in SD card. After flight, convert log file using the 'Cleanflight Blackbox-tools' to obtain the .GPX file. Open file in Google Earth Pro and try to slide bar in top left corner, you cannot, and the flight path remains faint. You cannot edit the times either. I believe this is caused by the flight start and end date/time both being the same (1/1/2020 1:00am (depending on timezone)).

[Screen shot here](https://scontent-dub4-1.xx.fbcdn.net/v/t1.0-9/118462780_10163932030875403_8198167309196457200_o.jpg?_nc_cat=101&_nc_sid=825194&_nc_ohc=hni_BLEzko0AX_LAgs2&_nc_ht=scontent-dub4-1.xx&oh=d92b3c2e670c6faf23cce3cfe0c6e054&oe=5F68E9AD)

## **Steps to Reproduce**
1. Obtain blackbox log file
2. Convert using Cleanflight Blackbox-tools
3. Open .GPX file in Google Earth Pro (File > Open)
4. Try to slide bar in top left corner, you cannot, and the flight path remains faint.

## **Expected behavior**
I should be able to slide the marker in the top left corner or press play and watch the flight. Flight path should become visible rather than faint, and it should be visible in 3D as per this video: https://youtu.be/GEGywqPD2Uk

## **Suggested solution(s)**
Make it record the time properly is my guess?

## **Additional context**
# version
# INAV/MATEKF765 2.4.0 Feb 11 2020 / 23:09:10 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

- FC Board name and vendor: Matek F765
- INAV version string: See above

stronnag, 2020-08-25:
Wrong repo, this a blackbox tools bug, that was fixed some time ago.
See https://github.com/iNavFlight/blackbox-tools/issues/12, where there are also fixed binaries.

closed 2020-08-25 by stronnag

AdamGFPV, 2020-08-25:
> Wrong repo, this a blackbox tools bug, that was fixed some time ago.
> See [iNavFlight/blackbox-tools#12](https://github.com/iNavFlight/blackbox-tools/issues/12), where there are also fixed binaries.

Thanks :)