iNavFlight/inav #6070: Crossfire nano doesnt work with Matek f411 wing

gliZZy-gamar, 2020-08-25:
## Current Behavior

## Steps to Reproduce

1. Connect crossfire to uart 1
2. Enable serial rx for uart 1
3. Mode is crsf

## Expected behavior

## Suggested solution(s)

## Additional context

---

shellixyz, 2020-08-26:
The inverter on this board is connected between the SBUS pad and RX1 but it should be possible to use the CRSF protocol with the receiver directly wired to RX1/TX1. Doesn't sound like an INAV bug though. Perhaps an electrical incompatibility

shellixyz, 2020-08-26:
I saw that there is another report of an issue with the F411W I will test. I have a wing using CRSF and F411W

Alex-Chist, 2020-09-12:
I have crossfire nano on F411 wing uart 1.
It works. Just set in Receiver tab:
Serial port inverted: Off
Serial receiver half-duplex: On

And it works.

flyhigh-hifi, 2020-09-26:
Same problem here. Matek should update their docs then if it's not compatible. They say directly to use UART1.

cmanley, 2020-12-01:
I've got the same hardware and it works for me. I've got the Crossfire Nano CRSF on UART1, GPS on UART2. Of course the GPS and RX only work when the board is powered by a battery and not just USB.
Unlike what Alex-Chist wrote: I left "Serial receiver half-duplex" set to Off otherwise it probably won't receive telemetry.
I'm not using the SBUS pad.
In Ports tab, UART1 has only "Serial RX" enabled, and UART2 has GPS selected below Sensors with baud 115200.

closed 2020-12-11 by DzikuVx