iNavFlight/inav #6071: Fixed wing servos freeze on 2.4.0, 2.5.1 and 2.5.2

CaptKrasno, 2020-08-26:
## Current Behavior
All servo outputs freeze after rapidly moving the control sticks on the transmitter.  If connected via USB the servo positions continue to move on the output page of the GUI.   

This behavior has been seen in both my fully configured "flight ready" configuration and a stripped down simple pass-through configuration (after a board erase).

The board was flashed with 2.4.0, 2.5.1 and 2.5.2 and all versions produced the same behavior.

This Behavior seems not to depend on the arm state of the FC.

I have recorded a video of this happening:  https://youtu.be/_eiGqy5Dbpw

This issue has been previously reported, unresolved and closed for 2.4.0 https://github.com/iNavFlight/inav/issues/5433

## Steps to Reproduce

1.  Switch FC to airplane mode
2.  Setup all channels in the mixer window and apply to servo outputs 3-10
3.  Rapidly move sticks for 10-30 sec.   
4.  all servo channels will freeze in position
5.  Sometimes if the sticks are continually moved the unit will un-freeze after some time.  

## Expected behavior
Servo output should follow the commands of the 

## Suggested solution(s)
Unknown.

## Additional context
I am using a Matek 765 Wing.   The servo outputs are wired DIRECTLY to the 2S LiPo battery and not through the onboard BEC.  I have an analog airspeed sensor, gps/compass, taranis QX7 and frsky 8xr attached.

I have not completed the compass and accelerometer calibration yet.

Stripped down config (shown in video):  https://pastebin.com/hibswMQR

---
version 
INAV/MATEKF765 2.5.2 Aug  4 2020 / 10:30:32 (faaedc7cb)
GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

- FC Board name and vendor:  Matek 765 Wing
- INAV version string: GetFPV

digitalentity, 2020-08-27:
Connect the display and see what's shown on the output. It's very likely that you are entering the CMS.

CaptKrasno, 2020-08-27:
You were right.  I was entering the CMS.   I wasn't planning on using FPV on this plane (really only wanted failsafe RTH) so I was unable to see the menu coming up.

Thanks for the help.  

For reference to others with this problem: CMS is activated by setting the throttle to mid, yaw left, pitch up.   This was getting randomly activated when rapidly moving the sticks around.

closed 2020-08-27 by CaptKrasno