iNavFlight/inav #6073: Throttle deadband in navigation modes on FW

avs2, 2020-08-27:
## Current Behavior

In the navigation modes on FW, throttle is managed according to Throttle = `nav_fw_cruise_throttle` - (`nav_fw_pitch2thr` * `pitch_angle`), where negative `pitch_angle` means pitching up. When external disturbances cause the aircraft to gain or lose altitude, INAV will pitch up and down in order to get back to the set altitude. This causes the throttle level to fluctuate which is visible in the OSD and you can also hear this if the aircraft is nearby, or on your HD video. You can tune out most of the fluctuations by setting the correct `nav_fw_pitch2thr` and altitude PID's, but some fluctuations are unavoidable. This gives off a bit of a nervous impression of what the aircraft is doing. 

## Desired Behavior

Throttle stays at `nav_fw_cruise_throttle`, even when small pitch corrections are made to maintain altitude. When `nav_fw_cruise_throttle` is set to a safe value pitching up a few degrees should not cause a stall.

## Suggested Solution

I think the easiest solution would be to implement some sort of (configurable) deadband around zero degrees pitch in which throttle is not increased or decreased. Outside of this deadband, throttle would increase/decrease for any additional pitch up/down degrees.

For example, with the current default values, when pitch is between +3 and -3 degrees, throttle would be 40%. When the pitch angle is up 4 degrees, throttle would increase to `40% - (-4° - -3°) * 1% = 41%`. When the airplane is pitched down 5 degrees, throttle would be `40% - (5°-3°) * 1% = 38%`.

Alternatively, there could be some other form of smoothing applied to the throttle command in the navigation modes. Either way, it would be nice to calm down the throttle response in the navigation modes.

## Who does this impact? Who is this for?

All fixed wing pilots.

## Additional context

I think this should be very easy to implement but I am not familiar enough with C to do a pull request myself.

Airwide, 2020-09-03:
@avs2 - Can you provide a cli dump and a blackbox log that explains the behaviour?

avs2, 2020-09-04:
I don't have those at hand, but I'm not sure how that could clarify my feature request. All I am suggesting is a small deadband around zero degrees pitch in which throttle is not changed.

Airwide, 2020-09-04:
@avs2 - I just wanted to see the extent of the oscillations, and rule out any configuration issues.

Airwide, 2020-09-06:
I have created a branch for this issue and will test it today. I’ll create a pull request if it’s successful.

avsaase, 2020-11-08:
Not stale

avsaase, 2020-12-08:
Solved by #6104.

closed 2020-12-08 by avsaase