iNavFlight/inav #6075: Blank map on mission controle

dmjacob2, 2020-08-27:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

I have looked for this issue and have not found in after searching.  The map in mission planner.  The GPS map on the GPS screen works fine.  So, very confusing.  I have tried to switch to bing with an API key and it works also in the GPS map, but not on mission planner.  

## Current Behavior
Black screen where the map should be on mission planner.  Does show the words Atlantic ocean.  It also allows me to create waypoints; so everything looks fine except the map is black.

## Steps to Reproduce

1.Open inav
2.Connect working drone
3.Go to mission planner section
4.No map

## Expected behavior

Should be a map just like the GPS screan
## Suggested solution(s)

this is a bug I thing, so fix bug
## Additional context

---

Entering CLI Mode, type 'exit' to return, or 'help'

# version
# INAV/KAKUTEF7HDV 2.5.2 Aug  4 2020 / 09:47:08 (faaedc7cb)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

Here is a picture of the screen
[image]

- FC Board name and vendor:
- INAV version string:

stronnag, 2020-08-27:
Wrong repo, this should be in the configurator repo, where it would be addressed if https://github.com/iNavFlight/inav-configurator/pull/1029 were every merged.

dmjacob2, 2020-08-27:
It seems to be working now.  Not sure what happened.  But thanks.

closed 2020-08-27 by dmjacob2

12ritchie12, 2021-07-21:
Just got same problem on inav 3.0.1