iNavFlight/inav #6081: accel calibration problem , omnibus fireworks v2 , inav 2.5.1  ,  2.5.2

igor-kutuzov, 2020-08-29:
omnibus fireworks v2 , inav 2.5.1 2.5.2 accel calibration is not working, 2.5.0 it's working well , also Current Sensor Offset in millivolt steps 32,6 is wrong , right is around 330 i think

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

digitalentity, 2020-08-29:
Please provide a more detailed description and a video demonstrating the acc calibration issue on a board flashed with full erase.

igor-kutuzov, 2020-08-30:
video https://youtu.be/hG2EbpWF8qQ

digitalentity, 2020-08-31:
This is not calibration - this is gyro/acc not detected. What board is this exactly (which model, manufacturer)?

igor-kutuzov, 2020-08-31:
omnibus fireworks v2 original, inav 2.5.0.  it's working well , 2.5.1   2.5.2. not working  
[image]

digitalentity, 2020-08-31:
This may be caused by the workaround for Omnibus F4V6. Try going to CLI and executing the following:

```
set gyro_to_use = 1
save
```

igor-kutuzov, 2020-09-01:
it's working now, thanks.

closed 2020-09-01 by digitalentity