iNavFlight/inav #6084: INAV FixedWing Omnibus F4V3 Arming not possible with Mission

FXvBH, 2020-08-31:
Hello

What is working:
The FX-86 Phantom flies perfectly.
Manual mode and horizonmode works perfectly. 
RTL (RTH) works, no problem.
Arming with the switch is no problem when I have enough sats.
When arming is oke, I can run the motor.

Planning a route with mission planner is going well.
If I check this with the laptop, I can simply retrieve the mission from the EEPROM.
With the sticks I upload from the EEPROM to the FC for before arming.

But then there is a problem.
After uploading with the sticks I can no longer arming the motor,  the motor will not start.

What's going wrong?

Thanks, Frans

stronnag, 2020-08-31:
posting the mission file and arming location would help ....

perhaps you are not within `nav_wp_safe_distance` of the 1st waypoint. 

See also https://github.com/iNavFlight/inav/wiki/%22Something%22-is-disabled----Reasons

FXvBH, 2020-08-31:
I have a mission with 3 WayPoints as start for testing a mission

Before stick command

Entering CLI Mode, type 'exit' to return, or 'help'

# wp
#wp 0 invalid
wp 0 0 0 0 0 0 0 0 0
wp 1 0 0 0 0 0 0 0 0
wp 2 0 0 0 0 0 0 0 0
wp 3 0 0 0 0 0 0 0 0
wp 4 0 0 0 0 0 0 0 0
wp 5 0 0 0 0 0 0 0 0
wp 6 0 0 0 0 0 0 0 0
wp 7 0 0 0 0 0 0 0 0
wp 8 0 0 0 0 0 0 0 0
wp 9 0 0 0 0 0 0 0 0
wp 10 0 0 0 0 0 0 0 0
wp 11 0 0 0 0 0 0 0 0
wp 12 0 0 0 0 0 0 0 0
wp 13 0 0 0 0 0 0 0 0
wp 14 0 0 0 0 0 0 0 0
wp 15 0 0 0 0 0 0 0 0
wp 16 0 0 0 0 0 0 0 0

After stick command

# wp
#wp 3 valid
wp 0 1 521627054 44132530 8000 0 0 0 0
wp 1 1 521623368 44095623 8000 0 0 0 0
wp 2 1 521625211 44132530 8000 0 0 0 165
wp 3 0 0 0 0 0 0 0 0
wp 4 0 0 0 0 0 0 0 0
wp 5 0 0 0 0 0 0 0 0
wp 6 0 0 0 0 0 0 0 0
wp 7 0 0 0 0 0 0 0 0
wp 8 0 0 0 0 0 0 0 0
wp 9 0 0 0 0 0 0 0 0
wp 10 0 0 0 0 0 0 0 0
wp 11 0 0 0 0 0 0 0 0
wp 12 0 0 0 0 0 0 0 0
wp 13 0 0 0 0 0 0 0 0
wp 14 0 0 0 0 0 0 0 0
wp 15 0 0 0 0 0 0 0 0
wp 16 0 0 0 0 0 0 0 0
wp 17 0 0 0 0 0 0 0 0
wp 18 0 0 0 0 0 0 0 0
wp 19 0 0 0 0 0 0 0 0
wp 20 0 0 0 0 0 0 0 0

FXvBH, 2020-08-31:
After the stickCommand it isn't more possible to run the motor.
Without stick command it is possible to start (arming) the motor.
I was close to the first waypoint.
But is it necessary for arming with mission to nearby the first waypoint?
If you are to far fromfirst WP then it isn't possible to activate the mission.

What for commands I must type in the CLI for info to you?

Grt, Frans

stronnag, 2020-08-31:
The home location when you attempt to arm. "I was close" is neither useful nor analysable. Any of the four corners of the "obvious" field are further than the default save distance from WP1 (see example, home is assumed to be the brown icon).
A cli `diff` might also help.
[image]

FXvBH, 2020-08-31:
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# INAV/OMNIBUSF4V3_S5S6_SS 2.5.0 Jun  7 2020 / 11:25:59 (694767938)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 -80 0 -1
smix 1 3 1 -80 0 -1
smix 2 4 0 -80 0 -1
smix 3 4 1 80 0 -1

# servo

# logic

# gvar

# gf

# feature
feature -TX_PROF_SEL
feature -AIRMODE
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 2 115200 115200 0 115200
serial 5 64 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 900 1075
aux 1 2 0 1725 1900
aux 2 10 0 925 1100
aux 3 28 0 1500 1675
aux 4 12 0 1925 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 4 7 V
osd_layout 0 1 13 1 H
osd_layout 0 7 12 11 V
osd_layout 0 8 17 2 V
osd_layout 0 13 22 5 V
osd_layout 0 14 3 9 V
osd_layout 0 15 23 11 V
osd_layout 0 20 2 13 V
osd_layout 0 21 2 14 V
osd_layout 0 22 13 2 V
osd_layout 0 23 2 5 V
osd_layout 0 28 23 6 V
osd_layout 0 30 2 12 V
osd_layout 0 48 23 7 V
osd_layout 0 49 23 8 V

# master
set gyro_hardware_lpf = 20HZ
set acc_hardware = MPU6000
set acczero_x = 124
set acczero_y = -25
set acczero_z = -55
set accgain_x = 4097
set accgain_y = 4059
set accgain_z = 3985
set mag_hardware = NONE
set magzero_x = -178
set magzero_y = -166
set magzero_z = -43
set baro_hardware = NONE
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = SBUS
set failsafe_procedure = RTH
set align_board_pitch = 40
set platform_type = AIRPLANE
set model_preview_type = 8
set small_angle = 180
set applied_defaults = 3
set nav_rth_climb_first = OFF
set nav_rth_allow_landing = NEVER
set nav_rth_alt_mode = AT_LEAST_LINEAR_DESCENT
set nav_rth_altitude = 5000
set nav_fw_cruise_thr = 1700
set nav_fw_max_thr = 2000
set nav_fw_dive_angle = 20
set osd_video_system = PAL
set osd_dist_alarm = 750
set name = FX-Phantom

# profile
profile 1

set fw_p_pitch = 0
set fw_i_pitch = 15
set fw_ff_pitch = 70
set fw_p_roll = 0
set fw_i_roll = 15
set fw_ff_roll = 30
set fw_p_yaw = 0
set yaw_lpf_hz = 30
set tpa_rate = 33
set tpa_breakpoint = 1300
set roll_rate = 25
set pitch_rate = 13
set yaw_rate = 9

# battery_profile
battery_profile 1

# end the command batch
batch end

FXvBH, 2020-08-31:
The brown icon is where I am connect the battery with the plane and where I try to arm including the waypoints

FXvBH, 2020-08-31:
If I understand correctly, the arming of a mission must take place within the minimum distance of WayPoint 1, because otherwise arming including mission will not work?
In that case I may have been just too far away?
For so far I can see is it 100 m
set nav_wp_safe_distance = 10000
That is 100 meters.

FXvBH, 2020-08-31:
I have set safe WP distance to 400 meters.
Then another mission from my garden home to the golf course and back.
After loading the mission from EEPROM to FC board, then arming, I can start the engine with a tap of gas and immediately shut it down.
I'll try again tomorrow morning, now dark.
I will let you know if it works.

Grt, Frans

FXvBH, 2020-09-01:
Hello Stronnag

posting the mission file and arming location would help ....

perhaps you are not within nav_wp_safe_distance of the 1st waypoint.

These sentences shook me awake.
This is the first time that I do anything with an FC and Fixed Wing.
It seems to be working now.

What have I done:
set nav_wp_safe_distance = 40000

I think at safe distance 100 meters I was just too far to start.

But the cruise speed from 1700 was too fast, and the mission too short to take the turn properly.
But something is happening.
Tomorrow another test, bigger mission, with the points further apart.
And the cruise speed to 1500.
If everything goes well tomorrow, I will set the rest on the field.

Maybe tuning the cruise speed to 1400 or 1450, if it works tomorrow, then I am very happy.

Thanks, Frans

FXvBH, 2020-09-03:
It's oke now.
The OSD give the text
TO WP
The plane follows the WP's.
Thanks, Frans.

stronnag, 2020-09-03:
Closing. In future, it would be appreciated if you could please use the social media channels for support.

closed 2020-09-03 by stronnag