iNavFlight/inav #6085: momentary disarms and flight mode changes in flight

jetstream123, 2020-08-31:
****

## Current Behavior

I have noticed that my plane always disarms, arms and changes flight modes when im flying. It happens every 9 seconds and only last about 30 milliseconds. On osd i can see flight mode change and feel the change in flight mode. Im using frsky L9R receiver and one of my friends having the same issue with r9 system. Also i changed arm switchs position. It is arming at down position but plane is still disarming. 
[image]

here is the dvr, look at the flight mode you can see its changing https://mega.nz/file/WPJmkYgQ#agu7xUEngf7Zfr_y6PO69PCgASsKL_tRVa1ZKikgwBE

and here is the log
https://mega.nz/file/zeg0GCrQ#cQV5G5XakT3ibw5vslYsDLcpJhnJ7cD1SBDCnHk-N80

## Expected behavior

It shouldnt change flight modes.

## Additional context

dump
https://pastebin.com/0LahLPjx
---

- FC Board name and vendor:
- INAV version string:
# INAV/MATEKF405SE 2.5.1 Jun 16 2020 / 12:50:26 (098a574b3)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

DzikuVx, 2020-09-01:
Which failsafe scenario do you have configured in OpenTX? ?

jetstream123, 2020-09-01:
You can't change failsafe option on transmitter with L9R receiver, so i cant check it right now and i have set it long time ago, if im not mistaken it is set to no pulses.

b14ckyy, 2020-09-07:
Seems to have something to do with the random disarms and mode changes during FrSky failsafes. All protocols and systems we have tested where affected by this. FrSky receivers seem to output RX signal recovered packed sometimes before all 16 channels are fully recoverd.

jetstream123, 2020-09-08:
> Seems to have something to do with the random disarms and mode changes during FrSky failsafes. All protocols and systems we have tested where affected by this. FrSky receivers seem to output RX signal recovered packed sometimes before all 16 channels are fully recoverd.

@b14ckyy so is there any solution?

b14ckyy, 2020-09-08:
This should be fixed by #6088 I hope.
Darren Lines made a detailed video about it. Maybe @digitalentity can confirm if this is what his fix is about. https://www.youtube.com/watch?v=rIFTbSQ2Wjc

The only reliable workaround we found so far is, to reverse the ARM channel and make it armed on 1000 and disarmed on 2000. But this will only prevent disarming, Not the random mode changes that can happen for a few ms during a failsafe. 

This also prevents us to use the failsafe during mission = OFF feature reliably as this glitch causing INAV to drop the mission progress. So we need to wait for the next release to make a proper workaround or FrSky fixes that on the receiver firmware (sigh)

jetstream123, 2020-09-08:
@b14ckyy i tried reversing the arm switch as its shown in the video, its arming at 1000 but same thing still happens. Hope it will be fixed with #6088 because i dont think frsky will fix it.

b14ckyy, 2020-09-09:
> @b14ckyy i tried reversing the arm switch as its shown in the video, its arming at 1000 but same thing still happens. Hope it will be fixed with #6088 because i dont think frsky will fix it.

OK I just watched the DVR. Your RSSI was high so that is a different topic. Has nothing to do with the glitch explained in the video from Darren.

Airwide, 2020-09-15:
I had a disarm this weekend during RTH when testing #6104. The statistics showed disarmed by switch. It sounds like the FrSky bug @b14ckyy references above. What's worrying is that #6088 was already merged when I created the branch in pull request #6104, which means it didn't fix the problem.

If I can provide further insight to the problem, I'll be happy to. I just don't know how.

[arwing_disarm.mov.zip](https://github.com/iNavFlight/inav/files/5227265/arwing_disarm.mov.zip)
[LOG00002.TXT.zip](https://github.com/iNavFlight/inav/files/5227267/LOG00002.TXT.zip)

thesingerdenelle, 2020-09-20:
Only just stumbled upon this. I literally just lost a plane over a beach today because of this very bug!

b14ckyy, 2020-09-21:
We made a blog post about that. And How to avoid it for now as a workaround. https://inavfixedwinggroup.com/rc-news/frsky-failsafe-disarm-bug-that-you-need-to-know-about/

FrSky is also looking into that at the moment. Let's hope they'll fix it soon but I still hope that INAV will also be "hardened" on that as it is easier for people to update their FCs than their radios and all receivers.

closed 2020-11-26 by DzikuVx