iNavFlight/inav #6093: Smart audio and led_strip not working

nangel0, 2020-09-03:
#5458  Current Behavior
Can't change vtx settings, or configure led_strip 

MATEK F411 Wing
vtx Akk ultimate x2 US version 
Single 2812b led light

## Steps to Reproduce
1. Flash latest inav version (2.5.2) from updated configurator. 
2. Try to change vtx and configure led
3. Can't change vtx (shows A :/ on OSD) and "A...... 144" in top of Vtx settings screen 
4.

## Expected behavior
Able to change vtx chanel and configure led

## Suggested solution(s)
Rolling back to 2.5.1 using. Hex file from matek site solved, so trying to compare codes may solve.

LeiShifu, 2020-09-04:
TBS smartsudio has the same problem on MATEK F722 and HGLRC F722
Vtx:FXT ARES 1W

**Steps to Reproduce**
1.Flash latest inav version (2.5.2) from updated configurator
2.use airplane default setting
3.set any hardware UART->Peripherals to TBS SmartAudio
4.set channel and power in the osd

**Problem behavior**
nomatter what channel/power level setted, TX pin can only output 
10010101-01101010-10101010-10000010-10000000-10000000-00101101-01101000-00000100 (I can't upload oscilloscope image)at 205us per bit(about 4.9kbps, MSB first),and the full segment signal last 15.1ms and output every 200ms(5Hz).
but the correct SmartAudio signal should lasts about 30ms (test with betaflight 4.0.0,MATEK F722,FXT ARES 1W).

**Expected behavior**
Able to change vtx power by TBS SmartAudio

DzikuVx, 2020-09-06:
SS2 and LED on this board are the same MCU pin (PA8) and it's just physically not possible to  use both functions in the same time

closed 2020-09-06 by DzikuVx