iNavFlight/inav #6094: INAV development IDE

johncook1212, 2020-09-03:
Hello to all,

I work on a project with a small drone and would like to code a few parts into INAV, hence I've read the "build with eclipse" document but could ramp up an appropriate IDE. also tried with older versions of eclipse but it didn't work.
Could someone please share a more up to date the required environment? and some short steps?

Thanks in advance,
John

DzikuVx, 2020-09-03:
> issue, with a confidence of 0.76. Please mark this comment with 👍 or 👎

Just use Visual Studio Code with WSL plugin like most of devs

johncook1212, 2020-09-03:
@DzikuVx  - Thanks for the prompt answer. 
As I am new to FW development, are there other packages or toolchains that are required to start?
One of the target is to work on the places where PID is coded. So if there is more info regarding on where to dig it will be great :)
Best,
John

DzikuVx, 2020-09-03:
Full text search is your friend when finding what goes where. Surprisinly PID code is in the pic.c file.

On topic of VS Code, C/C++ plugin from Microsoft and Remote - WSL plugis (also Microsoft) is all you need for starters

closed 2020-09-03 by DzikuVx

johncook1212, 2020-09-04:
@DzikuVx - Thanks, I will continue from there.
John