iNavFlight/inav #6096: Battery profile not saving PID's

VoicOfReason, 2020-09-03:
I am flying a lightweight 7" quad which runs well on both 4s and 6s but ONLY if I use different PID's for each battery. I tried using the auto battery profile switcher and I got it detect which battery I am on, but, to my dismay, different PID settings for each battery profile don't save. If I change PIDs for one, it changes them for all. I can't seem to assign different profiles to battery profiles, so I am stuck. If this is not a defect, could different PID settings per battery profile be implemented? I doubt few people can use same PID's between 4s and 6s.

Using latest (non-beta) iNav 2.5 as of today's date.

DzikuVx, 2020-09-04:
Battery profile stores only battery information. To store different PID gains you have to use rate profile, not battery profile or both at once

closed 2020-09-04 by DzikuVx

VoicOfReason, 2020-09-04:
I see. Is there any way to currently to auto switch PID profiles based on battery cell count? If not, could this feature be added? As I said before, the PIDs that work at 4s don’t work at 6s as the quad literally either shoots into the sky or flips over in the air (on 6s with 4s pids).