iNavFlight/inav #6097: PwM init Error

Cutbangrian, 2020-09-03:
just get my nano talon with F411-wse.
Installed inav 2.5.0

At first everything is okay, Im doing a basic setting on my nano talon. Reversing the wrong movement of elevator and rudder since nano talon use v tail.

But suddenly the servo wont work, I thought I forgot plugged in the battery. But I wasn't.

Battery is still on and I confused.

So I opened the monitor to see the system message on osd.

And it says pwm init eror. 

What was wrong?

Never experienced this before

digitalentity, 2020-09-04:
Please make and share the configuration dump

Cutbangrian, 2020-09-04:
[image]

Cutbangrian, 2020-09-04:
I reflash it with 2.5.2 now.  So i dont have dump data anymore.

I stick to 2.5.0 since it reliable than the newer 2.5.2.

2.5.2 version is newer and we dont know what bug inside. So I choose 2.5.0

While I'm setting up, suddenly the servo is not moving.
I checked thru osd and it says pwm init error.

Couldn't find any explanation about that system message in inav wiki.

So I just flash the new fw 2.5.2.

Hopefully this problem not occur in 2.5.2 version😅

Thanks

closed 2020-09-06 by digitalentity

jayss66, 2024-10-09:
Its 2024 and I am having this problem.
Output mapping
Output (timer) | S1 (Timer 4) | S2 (Timer 4) | S3 (Timer 3) | S4 (Timer 3)
Missing servos 5-11
Seems like a hardware issue, but I don't know. _Speedybeef405mini ___
Cli dump
diff all

# version
# INAV/SPEEDYBEEF405MINI 7.1.2 Jun  1 2024 / 00:43:32 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# Timer overrides

# Outputs [servo]

# safehome

# Fixed Wing Approach

# features
feature SOFTSERIAL
feature TELEMETRY
feature RSSI_ADC
feature LED_STRIP
feature DASHBOARD
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map

# Ports

# LEDs

# LED color

# LED mode_color

# Modes [aux]

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# OSD: custom elements

# master
set gyro_zero_x = -9
set gyro_zero_y = -13
set gyro_zero_z = 6
set acc_hardware = ICM42605
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = DPS310
set applied_defaults = 1

# mixer_profile
mixer_profile 1

set platform_type = AIRPLANE
set has_flaps = ON
set model_preview_type = 34

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.300
mmix 1  1.000  0.000  0.000 -0.300

# Mixer: servo mixer
smix reset

smix 0 1 0 100 0 -1
smix 1 2 0 100 0 -1
smix 2 3 1 50 0 -1
smix 3 3 2 -50 0 -1
smix 4 4 1 -50 0 -1
smix 5 4 2 -50 0 -1

# mixer_profile
mixer_profile 2

# Mixer: motor mixer

# Mixer: servo mixer

# profile
profile 1

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
mixer_profile 1
profile 1
battery_profile 1

# save configuration
save

# 
Status check
# status
INAV/SPEEDYBEEF405MINI 7.1.2 Jun  1 2024 / 00:43:32 (4e1e59eb)
GCC-10.3.1 20210824 (release)
System Uptime: 359 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=ICM42605, ACC=ICM42605, BARO=DPS310
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
ESC Temperature(s): Motor Count = 2
ESC 0: 0°C, ESC 1: 0°C, 
Stack size: 6144, Stack address: 0x10010000, Heap available: 1828
I2C Errors: 0, config size: 10467, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = none, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 6, cycle time: 1014, PID rate: 986, RX rate: 9, System rate: 9
Arming disabled flags: CAL ACC RX CLI PWMOUT
OSD: MAX7456 [30 x 16]
VTX: not detected
PWM output init error: Not enough servo outputs/timers

Any Ideas? Guess I could use it on a wing! lol

dgk675, 2024-11-25:
@jayss66 I just ran into the same problem. Your setup contains mixers for 5 Servos and 2 Motors , while your [target](https://github.com/iNavFlight/inav/wiki/Boards,-Targets-and-PWM-allocations#board-speedybeef405mini) has 4 pwm outputs. Remove the outputs that are not used in your mixer tab and the remaining ones should be able to initialise properly.

sensei-hacker, 2024-11-25:
The SpeedyBee Mini has four outputs. Your trying to use eight.

Maybe you have the SpeedyBee WING Mini, which has more outputs. If so, flash the WING target.