iNavFlight/inav #6098: DSHOT + F7 falls from sky

Xedos9er, 2020-09-04:
I switched to INAV after running this quad on BF for a long time.
Yesterday on my 3rd flight, the quad just crashed right after I took of and have done 2 rolls. I have attached the logfile from this flight, to me it looks like a desync on motor 3 ?? Am I right?
[LOG00014.TXT](https://github.com/iNavFlight/inav/files/5172736/LOG00014.TXT)

Setup I was running:
-Kakute F7 HDV
-BLHELI_32, 48khz, Demag High, Timing 20°
-2508 1200kv Motor
-2khz pid loop, 800khz I2C, DSHOT 300
-RPM Filter on

Throttle idle % is set to 7, wich is already higher than on BF. Never had issues with BF @ 4khz, DSHOT 300.
But on BF I was able to set Dshot min rpm, wich helps a lot to prevent desyncs.

DzikuVx, 2020-09-04:
From the log I see that the board never rebooted or stalled and problems were with ESC connectivity. What's interesting is that it happened immediately after punch out and it looks like coincidence or electrical problem due to poor wiring. Please check Multishot

INAV has throttle_idle set to relatively high value by default and that prevents desync as well.

Xedos9er, 2020-09-04:
> From the log I see that the board never rebooted or stalled and problems were with ESC connectivity. What's interesting is that it happened immediately after punch out and it looks like coincidence [...]

Hi Pawel,
thanks for the fast reply. Wiring is ok, I checked everything. But I noticed that motor 3 isn't running as freely as the other motors when spinning it by hand. It seems to have mechanical issues. Also notice the pitch wobble during a roll in the log, motor 3 has always to work harder. It also gets warmer than the other ones. This would also explain the nose pitching to the up left when Punchout, motor 3 doesn't accelerate as fast as the others. Seems like this mechanical issue leads to desync since motor 3 gets way to slow with 0 throttle on hard manouvers. I didn't notice this behaviour on BF since min_rpm (Wich is not available in INAV) compensates this issue. As far as I can remember, I always had this nose pitching in BF Wich never could be tuned out. Hope that makes sense what I discovered.
I would run multishot, but it doesn't support rpm filters.

DzikuVx, 2020-09-09:
Also verify Blackbox logging rate. If it's enabled with 1/1 it might be the reason

Xedos9er, 2020-09-09:
Yes, logging rate is at 1/1. What would you suggest

DzikuVx, 2020-09-10:
Set logging rate 1/2 or 1/4. SD Card operations are taking too much time and MCU has no time to process everything else.
On my hand, I will as a conditions that lowers the logging rate to safer levels

PR #6127 (2020-09-16): Do not allow for Blackbox to work faster than 1kHz
INAV logs a lot of data into the blackbox. Due to that, running BB faster than 1kHz can cause falling from the sky as a worst case scenario.

Closes #6098
closed 2020-09-16 by DzikuVx