iNavFlight/inav #6107: Burn local INAV FW version

johncook1212, 2020-09-07:
Hello to all,

I'm taking first steps on developing inside INAV using VS CODE + WSL plugins (thanks to @DzikuVx from previous issue).
After that I've bumped into a problem with burning the local FW. The steps are as following:
1. Burning KakuteF7MINI board with online stable FW -> the board was connected via COM port as expected.
2.  According to this document "Building in Windows 10 with Linux Subsystem" (link below) - I've created a local FW by pressing "make TARGET=KAKUTEF7MINI" -> A hex file was created at "obj" folder.
3. I've burned the local FW by choosing the generated HEX file.
4. When re-opening "inav configurator" I saw that the board is recognized (COM port) but entered automatically to CLI mode.
I saw this message - (attached picture)...
[image]

My question is - how to change the version to be supported with the configuration?

Thanks to all
John

Link to document from step "2": https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%2010%20with%20Linux%20Subsystem.md

johncook1212, 2020-09-07:
Hi,
I've managed to work it out by manually changing the version of the FW (surprisingly version.h),
just a quick one more question - on both expeirments to flash a local FW, after the FW was flashed, the configuratior tool raised n error regarding connection with the bootloader. Disconnecting the FC and re-connection to the USB port solved it and I could start work with it.
This error message is known?
John

DzikuVx, 2020-09-07:
To configure INAV 2.6 you also need local copy of Configurator. Using official release might have side effects

closed 2020-09-07 by DzikuVx