iNavFlight/inav #6109: NAV AltHold / NavPosHold triggers wide open throttle

danhouldsworth, 2020-09-07:
Hi 

So I've recently completed a dedicated multi-rotor build for iNAV. My motivation for this build being about experimenting with inertial sensor fusion using a full suite of available sensors. 

Hardware  : 

* FC : Matek F722SE / Flashed with iNAV 2.5.2
* ESC : HobbyWing XRotor 4in1 60A / Flashed with BLheli32 Rev32.7
* Frame : Armattan CF 226
* Motors : Emax 2205 2300kv
* Props : Gemfan 4023
* Power : Turnegy Graphene 1300mAh / 4s
* Filtering Electrical : Rubycon ZLH 35V/470uF
* Filtering Physical : Silicone grommets M4 to M3 isolating FC 

Sensors :

* IMU : MPU6000 (onboard) / SPI
* Baro : (onboard)  / I2C 800kHz
* Mag : (on GPS unit) / I2C 800kHz
* GPS : Beitian BN-880 / USART @ 38400
* Optics : Matek Optic Flow & Lidar Sensor /  USART MSP
* RX : Spektrum Sat / USART @ 115200

Configuration

* ESC protocol : DSHOT600
* Loop time : 2kHz

I've tuned the PIDs so in both Rates (Acro/Gyro) mode or Horizon (Accel) mode it flies well with good control authority and with no visible or audible oscillations even under high power. Hovering at ~50% throttle stick.

Mag & Optics have been calibrated using the iNAV configurator. Running several times in various locations to check for anomalies.
GPS has a good & correct 3D fix

Configurator Sensors tab show that optic flow giving correct readings when between 2-200cm.

However, when I engage either AltHold or PosHold modes (whether Surface mode is engaged or not) it appears to go into wide open throttle.
I've reduced the Alt hold PIDs to near zero and still get the same result.

Could someone advise if I've misunderstood these Alt Hold / Pos Hold modes? 
Or if there's anything obvious in the setup I've missed?

I'm expecting it to attempt to hold the Alt / POS at the point that I engage that mode.

[image]
[image]
[image]

digitalentity, 2020-09-07:
Please provide configuration + logs of the issue

danhouldsworth, 2020-09-08:
Thanks @digitalentity - 
Let me just double check configuration = screen shot of iNAV Configurator > PIDs / Rates / Feedforward values?
For the logs - (sorry for the additional question) - does the F722SE have on board flash other than the SD card slot?
If not I will source a Class 10 SD card mentioned in #2796 and then come back

digitalentity, 2020-09-08:
By config I mean - go to CLI, execute `dump`, copy the output and share it via pastebin or as a file. 
Re: logs - only SD-Card logging is possible on that board.

FreeLiveTutor, 2020-09-08:
Config dump attached:
[F722SE_INAV_CLI_DUMP.txt](https://github.com/iNavFlight/inav/files/5190506/F722SE_INAV_CLI_DUMP.txt)

A2BELE, 2020-10-23:
What did you find? Mine is doing same thing. Also F722se

closed 2021-07-13 by DzikuVx