iNavFlight/inav #6110: Fixed Wing Auto Throttle Deadzone for better Energy efficiency

b14ckyy, 2020-09-07:
## Current Behavior
If the plane's pitch angle changes slightly, the Auto Throttle system will constantly change the throttle calue for each degree pitch. 

## Desired Behavior
Having a Deadzone that allows pitch to change without adapting the throttle, would make the current consumption more stable and allows better estimation of flight times. Usually a good cruise throttle value should allow slight pitch up angles of 2-4° without the need to raise or lower the throttle. Especially in slightly windy conditions when altitude needs constant corrections of +/- 1 meter or two. A Deadband by default of 3° would keep the throttle at a constant level This could be adjustable from 0-10°

## Suggested Solution
Creating a editable Autothrottle deadzone with a range of 0-10 degrees (value 0-100) Each degree above or below the deadzone will then add pitch2throttle ratio as set up in inav.

## Who does this impact? Who is this for?
Fixed Wing pilots who do a lot of long range or Waypoint missions and want to squeeze the highest possible efficiency out of their planes. 

## Additional context
None

snaewe, 2020-09-08:
Duplicate of #6073 ?

Airwide, 2020-09-08:
> Duplicate of #6073 ?

Yes! 

@b14ckyy - See pull request #6104 .

b14ckyy, 2020-09-08:
@snaewe @Airwide yes guys you're right. Maybe used the wrong keywords when I looked for it. Will close here. thanks!

closed 2020-09-08 by b14ckyy