iNavFlight/inav #6113: "feature PWM_SERVO_DRIVER" no longer working from iNav 2.5.0

tomidland, 2020-09-08:
## Current Behavior
"**feature PWM_SERVO_DRIVER**" no longer working from iNAV 2.5.0
It works in iNav 2.4.0 though.

## Steps to Reproduce
1. Install iNav 2.5.0 or newer (newest stable now is iNav 2.5.2)
2. Enable "External PWM servo driver" in Configuration tab
3. Enter "**feature PWM_SERVO_DRIVER**" in CLI as instructed in the popup window
4. Then you get this"### ERROR: Invalid name"

## Expected behavior
I expected this to be enabled so I can use the PCA9685 servo driver.

## Suggested solution(s)
If you install iNav 2.4.0 it works. Was this feature removed or is there a different way to enable it now?

## Additional context
(I have tried to find an answer to this here on GitHub and on the facebook group. If I have missed the answer, please let me know.)

## Version 
INAV/OMNIBUSF4PRO 2.5.2 Aug  4 2020 / 10:01:31 (faaedc7cb)
 GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

stronnag, 2020-09-08:
https://github.com/iNavFlight/inav/releases/tag/2.5.0, in particular `servo_protocol` (new Settings)

tomidland, 2020-09-08:
Hum, I tried 
**set servo_protocol = SERVO_DRIVER**
but I still can not get any signal from the PCA9685.. (I also enabled it in the configuration tab)

tomidland, 2020-09-08:
[image]
I have a scope connected to it, and when I install 2.4.0 I get PWM signal.

gavinlowk, 2021-02-20:
The FC does not show up PWM_SERVO_DRIVER on CLI feature list ,
And I found that 
 " FEATURE_UNUSED_11 = 1 << 27,        // FEATURE_PWM_SERVO_DRIVER "         on config.h ( v2.6.0 ) is different from
 "  FEATURE_PWM_SERVO_DRIVER = 1 << 27, "     on config.h (v2.4.0)  

Is the External Servo PCA9685 driver cannot work due to INAV-configurator cannot set it to enable ?

gavinlowk, 2021-02-27:
I have spent a few days to read out the code and come up with below dirty fix below for this function to work.
Hope someone can make it work better later on.

1.You don't need to enable External PWM servo driver on configurator , since the FEATURE_PWM_SERVO_DRIVER 
has been changed to FEATURE_UNUSED_11 and seems to be enabled by default.

2. with CLI , run the command below to enable the servo driver
  servo_protocol=SERVO_DRIVER (I2C PCA9685 peripheral),

3. add the below code to fc_init.c , of course , you need to rebuild a hex( v2.6.0 ) and flash it to your own FC

#ifdef USE_I2C_IO_EXPANDER
    ioPortExpanderInit();
#endif
//----------add below
if ( servoConfig()->servo_protocol ==SERVO_TYPE_SERVO_DRIVER ){ pwmDriverInitialize(); }
//----------
    // Considering that the persistent reset reason is only used during init
    persistentObjectWrite(PERSISTENT_OBJECT_RESET_REASON, RESET_NONE);

    systemState |= SYSTEM_STATE_READY;
}

closed 2022-02-10 by DzikuVx