iNavFlight/inav #6119: Altitude estimation not correct in 3D Position hold mode while horizontal direction flying

ZlatanZ, 2020-09-10:
Dear Sirs,
I have a problem with altitude holding behaviour when running in 3D Position hold mode on my 17" quadcopter. When the quad is 3D position hold and I give the command for it to go forward or backwords the quad starts to go in the given direction but immidiately starts descending (not given by my commands - rcData[3], but visible on rcCommand[throttle] in log). It descends maximum around 3 meters at horizontal velocity of about 35kmh. When I am stoping the quad (horizontal velocity dropping to 0) the quad drops for another meter and then starts ascending and gains the lost 4 meters in next 5 seconds. You can see the behaviour on the video where I am flying forward and backwords accross the field.
In the blcakbox logs I don't see this big drop in altitude and I am not able to find the reason for this behaviour.

I have tried changing following till now but doesn't help:
 - parameters nav_mc_pos_z_p, nav_mc_vel_z_p, nav_mc_vel_z_i, nav_mc_vel_z_d in complete ranges from 5 to 200
 - inav_w_z_baro_p = 0 -> the quad doesn't keep altitude (no matter the inav_w_z_gps_p setting)
 - inav_w_z_baro_p = 0,01 -> the quad starts to slowly accelerating up or down (no matter the inav_w_z_gps_p setting)
 - inav_w_z_baro_p = 3 -> the quad starts osscilating in altitude aggressively(big amplitude - 10 meters) but slowly (few seconds)
 - inav_w_z_gps_p setting, inav_w_z_gps_v -> changing this value seems to have small or no effect on behavior

Build:
Custom 17x5.5" propellers quadcopter, 5008 360kV motors, FC Omnibus F4 v3 on the rubber dumpings. onboard bmp280 baro covered with foam.

I have attached the blackbox log, photos of the quadcopter, osd video of the behavior, cli diff parameters:
[image]
[image]
[LOG00270.TXT](https://github.com/iNavFlight/inav/files/5204207/LOG00270.TXT)
[INAV_cli_20200909_170856.txt](https://github.com/iNavFlight/inav/files/5204208/INAV_cli_20200909_170856.txt)
[0000001-2607-00.00.18.733-00.00.56.533.mp4.zip](https://github.com/iNavFlight/inav/files/5204280/0000001-2607-00.00.18.733-00.00.56.533.mp4.zip)

- FC Board name and vendor: tried on two boards I have: Omnibus F4 v3 and Omnibus F4 v3s
- INAV version string: INAV/OMNIBUSF4V3 2.5.2 Aug  4 2020 / 10:05:34 (faaedc7cb)

What am I doing wrong? Which parameters are influencing the altitude behavior? How is the altitude position estimated?
Why does it at the end slowly gain altitude? How to make it keep altitude when flying horizontally?

Best Regards,
Zlatan

digitalentity, 2020-09-11:
Put foam over the barometer and isolate it from direct airflow. GPS altitude is not used on multicopters. You can try completely disabling the barometer and enabling `inav_use_gps_no_baro` to use GPS altitude instead of the barometer as a reference.

ZlatanZ, 2020-09-12:
Thank you @digitalentity, but unfortunately that doesn't seem to solve my problem.

I tried now to protect the board with more foam and ducktaped the board around to protect the barometer from airflow. The airflow is defenetly not coming from the motors because they are to far away and spinning too slow. The airflow could only come when moving directionaly. But the funny thing is that the lowest altitude drop effect I feel without any the foam and with the F4 board turned upside down, but it is still about 3 meters.

Trying inav_use_gps_no_baro is not recgognized from the board:
[image]

The complete board temperature (seen in blackbox log) is slowly falling down when directional flying, because of the airflow around the board. I am not sure if this has any effect on the baro readings. **How can I put the raw baro readings as a debug variable?**

**Touching the MPU with finger or with something cold or warm seems to have a similar effect like I am expiriencing during flight (slow oscillations in altitude). It seems to me that the MPU temperature is not taken into account when calculating real acceleration?!?**
[image]

On the baro I don't see any such temperature-altitude relation which is also expected because in the code I see that there is temperature compensation.

**Thank you once more and what else could I try?**

ZlatanZ, 2021-01-18:
Problem is still not solved, issue still active... any help from anyone?

closed 2022-01-26 by DzikuVx