iNavFlight/inav #6120: No Hdop Telemetry sensor

Totals680, 2020-09-11:
When i discover sensors in my Taranis X9D+ SE i cant get hdop to show up i do get GPS, GPSspd, and Alt, i would like use hdop in the lua when i try to configure it currently in the lua the setting "GPS HDOP View" is maked with "--"

Specs 
matek F405 wingse
crossfire diversity
Matek SAM-M8Q GPS
Taranis x9d+se
full size crossfire tx

OptimusTi, 2020-09-11:
Hello @Totals680,

Get sat lock and post a picture of the Pilot screen.

Totals680, 2020-09-12:
Seems the signal bars are connected to the number of satalites
[image]

OptimusTi, 2020-09-12:
The signal bars are connected to the HDOP. Check the inav luaTelemetry docs.

Totals680, 2020-09-12:
So what would be the reason it doesn’t give me an option to change it to a numerical value as this would be more informative

OptimusTi, 2020-09-12:
That would be a question for the LuaTelemetry Gitahub

and maybe this?

"Options without selections have either been disabled by another option setting or are not available on your transmitter or telemetry protocol."

closed 2022-01-28 by DzikuVx