iNavFlight/inav #6125: POSHOLD oscilando no ar / POSHOLD oscillating in the air

renanseibert, 2020-09-14:
Hello,

My Quad (description below) flies well in ANGLE mode, flies reasonably well in ALTHOLD (with small variations of up to 1 meter in altitude, acceptable), but in POSHOLD mode it oscillates at low frequency. The compass is calibrated and correctly points home direction. I tested the RETURN TO HOME function and it works perfectly - except for oscillations when the quad stops in the air. The oscillation was quite accentuated, but after increasing the hover throttle a little, the GPS stabilized better, but it still oscillates at an angle of approximately 15 degrees, swinging from side to side. Does anyone have any idea what it could be?

Quad: (F450, Omnibus F4 V3.1, GPS Ublox M8N + compass HMC5883L)

_Olá,

Meu Quad (descrição abaixo) voa bem em modo ANGLE, voa razoavelmente bem em ALTHOLD (com pequenas variações de até 1 metro de altitude, aceitáveis), porém em modo POSHOLD ele oscila em baixa frequência. O compass está calibrado, tanto que aponta corretamente a direção de casa. Testei a função RETURN TO HOME e funciona perfeitamente - exceto as oscilções quando o quad para no ar. A oscilção era bem acentuada, porém depois de aumentar um pouco o hover throttle o GPS estabilizou melhor, porém ainda oscila a um ângulo de aproximadamente 15 graus, balançando de um lado para o outro. Alguém tem ideia do que pode ser?_

> 

Quad: (F450, Omnibus F4 V3.1, GPS Ublox M8N + compass HMC5883L)

**CLI Diff:**

diff

# version
# INAV/OMNIBUSF4V3 2.5.2 Aug  4 2020 / 10:05:34 (faaedc7cb)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# logic

# gvar

# gf

# feature
feature -AIRMODE
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 2 115200 115200 0 115200
serial 5 64 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1800 2100
aux 1 1 0 900 1950
aux 2 3 0 1050 1475
aux 3 11 0 1475 1900
aux 4 10 0 1975 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 25 2 H
osd_layout 0 1 24 1 V
osd_layout 0 5 23 8 V
osd_layout 0 6 23 9 V
osd_layout 0 7 13 13 V
osd_layout 0 8 9 1 H
osd_layout 0 9 1 4 V
osd_layout 0 13 1 9 V
osd_layout 0 14 1 8 V
osd_layout 0 15 1 2 V
osd_layout 0 22 14 12 V
osd_layout 0 23 1 10 V
osd_layout 0 28 23 11 H
osd_layout 0 30 1 14 V
osd_layout 0 32 24 2 V
osd_layout 0 96 1 1 V
osd_layout 0 99 27 1 H

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_type = PT1
set gyro_stage2_lowpass_hz = 120
set gyro_stage2_lowpass_type = PT1
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 150
set dynamic_gyro_notch_min_hz = 60
set acc_hardware = MPU6000
set acczero_x = 124
set acczero_y = 7
set acczero_z = -259
set accgain_x = 4089
set accgain_y = 4076
set accgain_z = 3961
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 128
set magzero_y = 81
set magzero_z = 52
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set min_check = 1050
set serialrx_provider = IBUS
set throttle_idle =  12.000
set failsafe_procedure = RTH
set model_preview_type = 3
set applied_defaults = 1
set gps_sbas_mode = AUTO
set mc_airmode_type = THROTTLE_THRESHOLD
set nav_extra_arming_safety = ON
set nav_rth_altitude = 1000
set nav_mc_hover_thr = 1575
set i2c_speed = 200KHZ

# profile
profile 1

set mc_p_pitch = 30
set mc_i_pitch = 10
set mc_d_pitch = 20
set mc_p_roll = 30
set mc_i_roll = 10
set mc_d_roll = 20
set mc_p_yaw = 60
set mc_i_yaw = 25
set dterm_lpf_hz = 60
set dterm_lpf_type = PT1
set dterm_lpf2_type = PT1
set use_dterm_fir_filter = OFF
set yaw_lpf_hz = 30
set dterm_setpoint_weight =  0.300
set heading_hold_rate_limit = 30
set mc_iterm_relax = RPY
set d_boost_factor =  1.500
set d_boost_max_at_acceleration =  5000.000
set d_boost_gyro_delta_lpf_hz = 50
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# battery_profile
battery_profile 1

# end the command batch
batch end

#

ZlatanZ, 2020-09-15:
Hi @renanseibert, 
is it oscillating vertical or horizontal? What is the frequency? Do you have some video? Do you have some Flight blackbox log?

BR,
Zlatan

renanseibert, 2020-09-15:
Hi, @ZlatanZ 
Oscillation occurs when the quad is stopped in the air. It is swaying to one side and the other on the axis, exactly the same as when the PID value of the PID is low (however I believe that the PID is good, because in ANGLE the quad flies well). I'll try to make a video later.
Thank you

renanseibert, 2020-09-15:
@ZlatanZ 

I made a video in POSHOLD mode. Link: [https://youtu.be/J0-NO56Inec](url)
After the video I increased the hover throttle and stabilized a little. Could this be the problem?
Thank you

ZlatanZ, 2020-09-16:
Hi @renanseibert,
Just from watching the video I would assume that your mc_i_roll and  mc_i_pitch are to high, maybe lowering them towards 1 helps. 
It would be much easier when you could provide blackbox log.

This could also help:
- try carefully adjusting the mc pitch and roll pid values with potentiometer in flight. 
- try increasing the roll_rate and pitch_rate to something like 1000. 
- Playing with the parameters mc_p_level, mc_i_level and mc_d_level.
- If the problem happens only in 3D hold position mode, try adjusting the pids in flight: nav_mc_vel_xy_p, nav_mc_vel_xy_i, nav_mc_vel_xy_d, nav_mc_pos_xy_p.

Do every change carefully with small steps to avoid crash!!!

renanseibert, 2020-09-16:
Hi,
Today I managed to solve the problem of stabilizing in POSHOLD mode. It was necessary to correct the navigation PID value. The D gain was too high and therefore the motors err very aggressively when correcting the xy position. POSHOLD now works perfectly.
Thank you very much, @ZlatanZ , without your help it would not be possible to solve the problem
 (Sorry for my english ...)

closed 2020-12-11 by DzikuVx

pDR8iLQ4go, 2022-03-06:
@renanseibert: Do you remember the values of your parameters? You wrote: "The D gain was too high...". Are you talking abiut parameter _nav_mc_vel_xy_d_?