iNavFlight/inav #6128: Different RTH options For Failsafe and Switched

breadoven, 2020-09-15:
## Current Behavior
Only 1 RTH setup is currently available triggered by Failsafe or switched by user.

## Desired Behavior
Ability to select different RTH setups for Failsafe and switched RTH, e.g. MAX + Landing for Failsafe ... CURRENT + No Landing for switched.

## Suggested Solution

## Who does this impact? Who is this for?
All users.

## Additional context
Switched RTH is often used in a more controlled way to simply bring the craft home as directly as possible so can be used without the need to climb first which wastes energy. Failsafe RTH however may occur where direct return isn't desirable, e.g. flying a WP mission in hilly areas where it would be preferable to climb first.

0crap, 2020-09-21:
This is already in iNav 2.5, read the documentation.

set nav_rth_allow_landing = FS_ONLY

breadoven, 2020-09-21:
Landing isn't an issue, switched RTH can be cancelled to stop it landing, It's climbing when not necessary that's the issue. It's the difference between making it back and not if the higher power needed to climb causes a depleted battery to sag excessively causing low voltage issues.

0crap, 2020-09-22:
You did have a look at this in the documentation?
https://github.com/iNavFlight/inav/wiki/Navigation-modes#rth-altitude-control-modes

breadoven, 2020-09-23:
Given that the issue is related to avoiding an undesired switched RTH climb would it be possible to override the preset RTH climb by using Alt Hold ?. So if you want to cancel any climb setup in RTH you select Alt Hold whilst RTH is active which will stop the climb at that point with RTH continuing as normal with that reset max altitude. Only negative is you may have Alt Hold set before RTH is selected inadvertently cancelling the RTH climb. However if this is an optional setting it shouldn't be a problem.

breadoven, 2021-03-12:
Closing as covered by https://github.com/iNavFlight/inav/pull/6179.

closed 2021-03-12 by breadoven