iNavFlight/inav #6129: Target for STM32F411 development boards with STM32F411CEU6

HristoB, 2020-09-15:
## Current Behavior
Hi all. I've built and flown my own DIY flight controller based on STM32F103 development board, the so called "blue pill" but with flashed with cleanflight, the flashing and configuration is done via FTDI. I have managed to flash Betaflight stm32f411 target on onestm32f411 via usb in DFU mode as well as with stlink v2. The problem I have is that I cannot connect to the board on any of the UARTs with a FTDI as well as the USB port to complete the configuration.
## Desired Behavior
Could someone help and point me to the right direction for building a target for STM32F411 development boards with enabled MSP on any UART so can be configured with a FTDI adapter. 
## Suggested Solution
MSP enabled on UART2 or 3, I'd like to contribute to the iNav project with compiling a target, testing and documentation regarding DIY FC based on these development boards and built with readily available components.

## Who does this impact? Who is this for?
This would benefit anyone willing to go an extra step and build own flight controller

## Additional context

digitalentity, 2020-09-17:
Adding support for a dev board seems a good first step of getting familiar with the code and build process in general for someone willing to contribute to the project.

As this is not a commercially available flight controller, use-cases of this target would be pretty much limited to personal education and tinkering. Core developers likely would not dedicate their time for a such a request as there is marginal benefit the the project as a whole.

We would likely accept a target for a widely available development board (i.e. a board from ST-NUCLEO family).

closed 2020-09-17 by digitalentity

HristoB, 2020-09-17:
> Adding support for a dev board seems a good first step of getting familiar with the code and build process in general for someone willing to contribute to the project.
>
> As this is not a commercially [...]

The question was intended as a help(feature) request, (the bot changed it to ISSUE) if "some one would assist building the target",  Not building the target for Me, with pin mappings and port functions in particular, or at least(someone) point me to the section of the code where I can modify, build, test, and of course, share back to the community with schematics, documentation etc...  As far as the Nucleo goes, I don't see any any point as for a few more quid one can go and get a ready F405 based flight controller and play around.. I'd rather donate my spare cash to the iNav project than shifting it to the Far East to some cloners...
Thanks for giving it a thought anyway....
H

digitalentity, 2020-09-18:
A quick follow up for useful pointers:
 - the target-specific code is located exclusively in the "target" [directory](https://github.com/iNavFlight/inav/tree/master/src/main/target) under a target name. I.e. for [MATEKF405 target](https://github.com/iNavFlight/inav/tree/master/src/main/target/MATEKF405).
- if you are using a supported CPU and sensors, adding a new target is a rather straight forward process - create `target.h` to define the hardware peripherals and some of the features; create `target.c` with timer mappings; done.
 - adding exotic hardware is a more complicated process (see [the efforts](https://github.com/iNavFlight/inav/pull/5827) to add support for H7 CPU family for example)

rossrosh007, 2021-08-24:
> Thanks

Hello HristoB,

Have you progressed/overcome the configuration issue on the STM32F411CEU6 diy FC? 
I'm interested putting together an FC of the same. If you could share the resources (schematics, solutions, etc) that'd be helpful.

HristoB, 2021-09-09:
> > Thanks
> 
> Hello HristoB,
> 
> Have you progressed/overcome the configuration issue on the STM32F411CEU6 diy FC?
> I'm interested putting together an FC of the same. If you could share the resources (schematics, solutions, etc) that'd be helpful.

Hi,

not really. Got MPU9250 but coldnt get it to work on SPI. Managed to flash an Unified Betaflight target on my F411 and configure it from the CLI, but to be honest, I'm not experienced enough to build a iNav target as I'd prefer iNav over Betaflight and I'm used to it.
Unfortunately, had to abandon the project as I couldn't get any iNav F411 target to work on it. No much of documentation on those either. No point really, I got myself a couple of F405 FC's  from China to play around. Just coding is not my thing..

rizacelik, 2021-10-29:
Build a GPS supporting 6-axis drone with INAV using the STM32F411CEU6 Development board.
https://youtu.be/aMPMUwnm-AA

skimans, 2021-12-06:
@rizacelik you can add source file here ?

synersignart, 2022-07-20:
I would Support this atleast have 1 or 2 Target development board on F4 and F7
with a wide selection of Sensors possible 
I know Rizacelik did one from a  STM32F411CEU6 Development board but sensor selection is some how limited Wish he could add more.  

like MPU9250 AK8963 BMP280 and more newer sensors down the line.