iNavFlight/inav #6131: Adding an RC Switch in INAV on a Servo Output

Tinkerer1234, 2020-09-16:
Hello,

I posted this in the forum to see if this was just a Newb question, and I have not been able to resolve it.

Firstly:

Matek F722-PX
INAV 2.5

My issue is this:

On the Matek F722-PX, you have (8) motor outputs and (2) Servo outputs.  For my set-up, I wish to add an additional (2) servo outputs utilizing the extra (4) motor outputs. Matek provides Servos on Pads S9 and S10.  I want to add servos on S5 and S6 as well.

When trying to customize the mixer, I add (4) Servos and Save and reboot.  I only have servos on the S9 and S10 slots in the mixer.  They show up on the outputs but are not bound to any output pins on the FC.

How do I define the S5 and S6 motor outputs to Servos?

As an experiment, I started with a custom "Other" mixer.  This would allow me to have as many servos as I wanted, but only the first (2) slots would register for Motors, and of course, I need (4).

I am sure there is a work around for this.  I do not want to make a custom firmware, because, well, I would have to learn how.

Any help would be appreciated.

IVData, 2020-09-17:
FWIW I have servos on S3, 4, 5 on the F722-PX without any issue. Maybe post your diff.

Tinkerer1234, 2020-09-18:
> FWIW I have servos on S3, 4, 5 on the F722-PX without any issue. Maybe post your diff.

I can get that on a Fixed wing set up.

Timers are set up so that S3,4,5 and 6 are all on TIM3.  So, if you are doing a Multi-Copter, this would not work.  I need 3 and 4 to Be MC, not FC so they are Motors and not servos.

If I could get Ubuntu to load on my machine with out an error, I would have had this fixed by now.

Tinkerer1234, 2020-09-18:
Here is what I changed:

From:

    DEF_TIM(TIM2, CH3, PB10,  TIM_USE_MC_**MOTOR** | TIM_USE_FW_SERVO, 0, 0),   // S7   UP1-7   D(1, 1, 3)
    DEF_TIM(TIM2, CH4, PB11,  TIM_USE_MC_**MOTOR** | TIM_USE_FW_SERVO, **0, 1**),   // S8   UP1-7   D(1, 6, 3)

to:

    DEF_TIM(TIM2, CH3, PB10,  TIM_USE_MC_**SERVO** | TIM_USE_FW_SERVO, 0, 0),   // S7   UP1-7   D(1, 1, 3)
    DEF_TIM(TIM2, CH4, PB11,  TIM_USE_MC_**SERVO** | TIM_USE_FW_SERVO, **0, 1**),   // S8   UP1-7   D(1, 6, 3)

Testing it out tonight.  Thoughts?

Can someone tell me what the 0, 0) denotes a the end of each definition?  S8 is the only one that has 0, 1 and I am questioning if it needs to be different.

Tinkerer1234, 2020-09-21:
Verified this worked.

You can only add (2) servos to a Matek F722-PX with this method if you are building a quad.  S3, S4, S5 and S6 are all on the same timer, so unless you figure out a different timer sequence or run your Motors at a lower Hz, these are not switchable.

As it is, simply changing MOTOR to SERVO on S7 and S9 will allow you to have 4 servo available.

Another side note, as of writing this, the Inav Configurator is 2.5.0.  The Git for the master files is 2.6.0.  As such, the are not compatible.  Be sure to load the archived 2.5.*.  If you don't the GUI is not available for the custom firmware. (Yes, I found out the hardware Saturday Morning)

closed 2020-09-21 by Tinkerer1234