iNavFlight/inav #6132: Inav target request for HIFIONRC F722 pro

krizw, 2020-09-16:
## Current Behavior
I came across the flight controller on a recommendation by Dronemesh (youtube), it has a tremendous specification, and I expected an Inav target would soon appear. I have been unable to contact the manufacturer to request their assistance with an Inav target 

## Desired Behavior
An INAV target or assistance getting one

## Suggested Solution
Development of an INAV target.

## Who does this impact? Who is this for?
Those who have purchased this flight controller with INAV in mind.

## Additional context

krizw, 2020-11-09:
I have contacted a retailer in the UK (cheapdrone.co.uk), that sells this board, and they are interested in supporting my target request by sending a board to a developer, Where and who is willing to assist?

DzikuVx, 2020-11-10:
@digitalentity ?

kang2000, 2020-11-17:
I hope this board will be able to run with INAV too.

rob-riggen, 2020-11-18:
I have this board and it is very capable. I also would like to see an inav target for hifionrc f7 pro v2.

digitalentity, 2020-11-19:
Retailer in the UK (cheapdrone.co.uk) wants this board to be featured in our wiki as well which is not in scope of [the hardware policy](https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md). I guess that's a no-go, unfortunately, unless there is _another_ retailer or board manufacturer who's willing to follow the process.

stronnag, 2020-11-19:
However, there is ample precedent for the board to have an entry in the official project documentation ("Board - FOO.md"), which can include links to vendors (as happens for a number of Banggood marketed MATEK boards), if that is the vendor's concern.

If they want additional publicity (mention on the Configurator, and / or in the wik ie.g. [Supported Boards](https://github.com/iNavFlight/inav/wiki/Supported-boards) (which looks horribly outdated) then perhaps there could be a support / sponsorship arrangement, should be vendor wish to be a supporter.

digitalentity, 2020-11-19:
> If they want additional publicity (mention on the Configurator, and / or in the wik ie.g. Supported Boards (which looks horribly outdated) then perhaps there could be a support / sponsorship arrangement, should be vendor wish to be a supporter.

Yes, that's what I communicated to them. Featuring a vendor is orthogonal to adding support for a board and should be subject to a separate discussion and agreement.

krizw, 2020-11-25:
> Yes, that's what I communicated to them. Featuring a vendor is orthogonal to adding support for a board and should be subject to a separate discussion and agreement.

Is this the end, or have there been other conversations with the vendor? I agree with your point and expected some issues regarding links to the companies product page. I hope something can be arranged that suits everyone.

digitalentity, 2020-11-25:
There were no further email exchange. I guess the vendor is not interested in INAV supporting this board.

ltwin8, 2021-01-13:
Hello, just saw this issue. I have worked on support for HIFIONRC F722 (AIO and Pro I hope, since both use same unified target in BF).
#6489 Please take a look. I could not test myself, because my hardware is still I shipping

waldviertler, 2021-04-04:
Hello!

I just got the new HIFIONRC F722 Pro V2 and so I tried to flash inav_2.7.0_HIFIONRCF722.hex on it. I use the INAV Configurator 2.6.1.
But after a successful flashing it told me that I have to update the firmware - ... 
Only CLI was showing after that - and I had to go back to BF.

ltwin8, 2021-04-04:
You need the beta of the configurator, just a Moment, i will Compiler 2.6 for you, That should work with configurator 2.6

ltwin8, 2021-04-04:
[inav_2.6.0_HIFIONRCF722.hex.zip](https://github.com/iNavFlight/inav/files/6255222/inav_2.6.0_HIFIONRCF722.hex.zip)

waldviertler, 2021-04-04:
Thank you very much!
I try it right now.
It looks as it works, but I have to find out why the magnetometer (BN-880) is not recognized.

ltwin8, 2021-04-04:
we came across this issue before, the "internal" i2c where Baro is connected is not the same as i2c pads afaik

ltwin8, 2021-04-04:
you can try to solder to the i2c used by the barometer, however that requires steady hands

waldviertler, 2021-04-04:
In BF I also had no magnetometer at first, but then I used this command in CLI:
set mag_bustype = I2C
set mag_i2c_device = 1
set mag_i2c_address = 0

-- but that does not work with INAV.

Sorry, but I am not that good in soldering...

ltwin8, 2021-04-04:
okay, maybe you then have to set something in inav? like mag-type?
maybe go with low-ish i2c speed

waldviertler, 2021-04-04:
Like 100 or 400 khz I2C Speed?
I have tried that - bud it does not find the magnetometer ether.

Lagarda, 2021-04-06:
Hi to All,

I'm novice here. 
I bought recently the board FC HIFIONRC F7 PRO V2. Would you please tell me how to flash it? 
I tried to connect via rx/tx but not successfully. If i press boot button and use usb port the system stops to see it.

waldviertler, 2021-04-06:
You have to connect the FC to your computer via a USB cable. Then open the INAV Configurator - there should a new com appear in the right upper corner. (like COM8). Go to Firmware Flasher in the menu on the left side.
Use "Load Firmware [Local]" from bottom right to load inav_2.6.0_HIFIONRCF722.hex from Itwin8 (you have to unzip the file from inav_2.6.0_HIFIONRCF722.hex.zip. Then use "Flash Firmware" (It has to be in DFU-mode for flashing).
I hope this helps!

Lagarda, 2021-04-06:
> You have to connect the FC to your computer via a USB cable. Then open the INAV Configurator - there should a new com appear in the right upper corner. (like COM8). Go to Firmware Flasher in the [...]

Doesn't work like that. As soon as i switch to DFU the com port is lost, and it sees as unknown device

ltwin8, 2021-04-06:
> Doesn't work like that. As soon as i switch to DFU the com port is lost, and it sees as unknown device

try impulserc driver fixer

waldviertler, 2021-04-06:
Thats ok. In DFU mode there are no com ports. But it does not flash?

Lagarda, 2021-04-06:
> Thats ok. In DFU mode there are no com ports. But it does not flash?

No unfortunately it doesn't flash

waldviertler, 2021-04-06:
what happens when you dont flash - instead connect to a com? Do you see the CLI menu?

Lagarda, 2021-04-06:
> what happens when you dont flash - instead connect to a com? Do you see the CLI menu?

Yes, cli mode is available.
Initial flash is betaflight 4.2.0

It connects to COM25, as soon as I try flash it having Inav or Betaflight it turns to COM11 (local port) when in boot mode and the flashing process endless

Lagarda, 2021-04-06:
> what happens when you dont flash - instead connect to a com? Do you see the CLI menu?

I even tried to connect through rx/tx using usb to TTL adapter and  demostratorGUI but it says the board is read-only

Lagarda, 2021-04-06:
[image]
[image]

waldviertler, 2021-04-06:
What happens when you connect to USB then type `dfu` in CLI - does it connect in dfu mode?

Lagarda, 2021-04-06:
> What happens when you connect to USB then type `dfu` in CLI - does it connect in dfu mode?

It says no such command

waldviertler, 2021-04-06:
Hmmm. Also in Betaflight?

krizw, 2021-04-06:
I would recommend watching some YouTube videos, painless 360, search for flashing FC firmware.

Lagarda, 2021-04-06:
> Hmmm. Also in Betaflight?

yeap

Lagarda, 2021-04-06:
I have installed DFU (the impulseRC) but it conflicts with win7 kernel

Lagarda, 2021-04-06:
[image]
[image]
[image]
[image]
[image]

Lagarda, 2021-04-06:
I tried to rich erase via gui demonstrator, it reflects smth, but anyway i cant succeed

waldviertler, 2021-04-06:
Usually you have to flash within BF oder INAV. I don't even know if its working with the gui demostrator...
What Itwin8 mentioned was this driver fixer from impulserc:
[https://impulserc.blob.core.windows.net/utilities/ImpulseRC_Driver_Fixer.exe](url)

waldviertler, 2021-04-06:
[image]

Lagarda, 2021-04-06:
Solved
I finally managed with STM 32 CUBE programmer.
I have uploaded the inav 2.6.0 but i'm in cli mode only? What should i do then for normal gui?

Lagarda, 2021-04-07:
I have flashed my FC with 2.7.0. but it works only in CLI mode, do you know how to fix it?

Foxli212, 2021-04-14:
Добрый день с компосом еще не решили рроблему ?

waldviertler, 2021-04-14:
Unfortunately not. I'm back to Betaflight.

Foxli212, 2021-04-14:
Betaflight есть удержания позиции ?

Lagarda, 2021-04-14:
> Добрый день с компосом еще не решили проблему ?

Нет еще. Попробовал оба компаса (магнетометра) QMC and HMC не работают.

Lagarda, 2021-04-14:
ПОдскажите, на какой порт GPS вешать? Does anybody can tell me what port to connect GPS?

waldviertler, 2021-04-14:
With _port_ do you mean UART?

Lagarda, 2021-04-14:
> With _port_ do you mean UART?

yeap UART

waldviertler, 2021-04-14:
If you soldered on like me, its on UART5.
I have:
UART2 = Crossfire Receiver
UART4 = Bluetooth (MSP)
UART5 = GPS
UART6 = DJI HD (MSP)
[image]

Lagarda, 2021-04-14:
Thanks, for your help! Do you know if any special settings for GPS module needed? Like i have in CC3D - 33800, UBX only, 5Hz, etc

waldviertler, 2021-04-14:
No, I dont have used any special settings for my GPS module (BN-880). GPS has worked instantly - only the magnetometer does not work with INAV. 
Which GPS module do you use?

Lagarda, 2021-04-14:
> No, I dont have used any special settings for my GPS module (BN-880). GPS has worked instantly - only the magnetometer does not work with INAV.

What's in the code? I mean what i2c address is used? 

> Which GPS module do you use?

[image]

waldviertler, 2021-04-14:
Your module shows: VCC, RX, TX and GND. 

Module --- FlightController
VCC goes to 5V
RX goes to T5 (this is UART5)
TX goes to R5 (this is UART5)
GND goes to GND

I dont see SDA and SCL -- these go usually to the SDA and SCL pads on the flightcontroller and only they use i2c!

virtus2232323, 2021-05-19:
When flashing in hifion rc with mamba f722 firmware, a magnetometer worked for me, it is connected to UART 3. but there is a minus, the barometer stopped working .... can someone give the firmware or instructions on how to fight !!!! Thank you!!!!

Foxli212, 2021-06-07:
Поевилась прошивка ???

Lagarda, 2021-06-07:
> When flashing in hifion rc f722 with mamba f722 firmware, a magnetometer worked for me, it is connected to UART 3. but there is a minus, the barometer stopped working .... can someone give the firmware or instructions on how to fight !!!! Thank you!!!!

I cant flash with Mambas firmware, the FC hangs on and no any response.

Lagarda, 2021-06-07:
> Your module shows: VCC, RX, TX and GND.
> 
> Module --- FlightController
> VCC goes to 5V
> RX goes to T5 (this is UART5)
> TX goes to R5 (this is UART5)
> GND goes to GND
> 
> I dont see SDA and SCL -- these go usually to the SDA and SCL pads on the flightcontroller and only they use i2c!

Yes thanks! UART5 works fine. I even switched to beitian GPS module.

Lagarda, 2021-06-07:
> If you soldered on like me, its on UART5.
> I have:
> UART2 = Crossfire Receiver
> UART4 = Bluetooth (MSP)
> UART5 = GPS
> UART6 = DJI HD (MSP)
> [image]

Would you gimme more details how to connect FrSky LRS to UART2 (R9M slim +OTA receiver)?

waldviertler, 2021-06-07:
[image]
Should be like this.

Lagarda, 2021-06-10:
> You need the beta of the configurator, just a Moment, i will Compile 2.6 for you, That should work with configurator 2.6

Hi. May I ask you to make change in the main code and add I2C addresses like for HMC5885? And make one more compilation then?

[22 more replies]

closed 2022-05-07 by DzikuVx