iNavFlight/inav #6134: CLI `diff` locks up FC with GCC 9.2 compiled images

stronnag, 2020-09-17:
In master (2.6) branch, when compiled with arm-none-eabi-gcc 9.2.1 (or any gcc 9.x cross-compiler), using the CLI `diff` command locks up the FC after the `diff` exceeds a certain size (c. 1KB). This issue does not occur with gcc 10.2 compiled images.

Discovered in the repo https://github.com/avs2/inav / branch `test-branch-avs2`, reproducible on master as of 2020-09-17.

* When built with `arm-none-eabi-gcc` version 10.2.0, hex work with no problem
* When built with either the legacy sdk compiler (9.2.1) or later 9.x series (e.g. 9.3.1 also tested), then the following is observed.
  * After new settings are successfully entered and saved (e.g. restore a previously saved `diff`), then the FC will randomly lock up on issuing a subsequent `diff`
  * By random, the first subsequent `diff` may or may not lock up the FC but eventually a `diff` will cause lock up. When the FC does lock up, it does always stop at the same place

The lock up is independent of the terminal emulator, it happens with the configurator, with mwp's `cliterm` and with `picocom` at least. Reproduced on MATEKF411SE and WINGFC boards.

[RCG discussion and background](https://www.rcgroups.com/forums/showpost.php?p=45456437&postcount=26489)

For information / discussion.

Solved by using the later stable compiler, but this is not an option for many users.

stronnag, 2020-09-17:
Note that this is mitigated using legacy compilers by reverting to size optimisation:
```
make TARGET=WINGFC OPTIMIZE_FLAG_NORMAL=-Os OPTIMIZE_FLAG_SPEED=-Os
```
generates a working target again. Looks like a compiler issue.

DzikuVx, 2020-09-18:
We can not revert everything to -Os as it would break multirotor performance

stronnag, 2020-09-20:
With 0c716ebdc2, this disappears. cmake is really magical; from trivial tests it appears to solve #6134; how I love Heisenbugs.

stronnag, 2020-09-22:
Closing, as the subtly different new build environment "fixes" this issue

closed 2020-09-22 by stronnag