iNavFlight/inav #6139: Problem with building in Windows 10 with Linux Subsystem

romko1, 2020-09-21:
Hello,
I followed the instructions given here:
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%2010%20with%20Linux%20Subsystem.md

In step: make MATEKF405
[image]
Thank you for any good advice.

stronnag, 2020-09-21:
Perhaps you could provide us with some information to be able to help:

* The output of `uname -a`
* The output of `ruby --version`
* The output of `ls -l /bin/sh`
* The output of `echo $SHELL`

stronnag, 2020-09-21:
And also please confirm that you ran all the steps in Ubuntu, specifically cloning the repo (and you did not unpack a zip file in Windows).

romko1, 2020-09-21:
Of course, I used the git command (no zip file in Windows).
oh, sorry, here you go:
[image]

huhen, 2020-09-21:
I have a similar problem - the reason for the parenthesis in the path variable.
eg:
>echo $PATH
>... /mnt/c/Program\ Files\ (x86)/Common\ Files/Oracle/Java/javapath ...

[This](https://github.com/microsoft/WSL/issues/1493#issuecomment-417639271) solution helped me(reset the second bit, for example 15->13)

romko1, 2020-09-22:
Hello,
slight forward.
I added `PATH = "$ PATH: / usr / bin"` to the .bashrc file, I modified the registry: the Flags registry value should be changed from 7 to 5 as a result and got here:
[image]

romko1, 2020-09-23:
I came across an error why it gave me an error. The reason I do it all is to create my own firmware. So far, compilers have not minded overwriting files in the TARGET directory. The remedy is as follows: it is necessary to modify the given files directly, e.g. in Midnight Commander (MC).

I finally managed to create my own firmware. As stated in the instructions, it is FW v.2.6.0, unfortunately with the current inav configurator v.2.5.0 I can't read it. 
Please is there a newer inav configurator? Well thank you.

stronnag, 2020-09-24:
> Please is there a newer inav configurator? Well thank you.

pretty recent builds may be found at http://seyrsnys.myzen.co.uk/inav-configurator-next/

DzikuVx, 2020-09-24:
OK, as I experienced the problem by myself, I was able to confirm that this works:

1. apply https://github.com/microsoft/WSL/issues/1493#issuecomment-417639271
1. Restart WSL and preferably Windows as well
1. Run `cmake ..` from `build` folder again
1. `make TARGET` should be working

closed 2020-09-24 by DzikuVx

MrD-RC, 2020-11-01:
Just a note. This fix worked on my PC. But, on my laptop, the value in flags  is 15 and not 7. Setting it to 5 reset the WSL environment, as if I had unregistered it.

PR #6146 (2020-09-24): WSL error troubleshooting
closes #6139
PR #6258 (2020-11-05): Update Building in Windows 10 with Linux Subsystem.md
Added fix for WSL v2 and the wsl.conf option for the fix.

Extends fix for Fixes #6139 
Fixes #6238