iNavFlight/inav #6140: Need some advice on 7" 6S low KV tune

Xedos9er, 2020-09-21:
I had very bad mid throttle oscillations when I switched from DAL 7056C to HQ 7045 Biblade props. Props got perfectly balanced. I notice very bad D-Term and P-Term oscillation looking at the Blackbox logs. I tried to reduce P and D gain without luck. After switching to 48khz PWM it got much better but still persists sligtly. But on 48khz there´s much more wooble in zero throttle situations due to lower bracking power. Any advice how to ged rid of these oscillations? Are my PIDS just to high for this powerful 6S 2508 1200kv motor setup? Log attached
[LOG00032.TXT](https://github.com/iNavFlight/inav/files/5257035/LOG00032.TXT)

FelixvdDonk, 2020-09-22:
I don't have a solution for you but I want to point out the issues you experience can be caused by the ESC, motor ,and Voltage choice combination. I have had issues with motors beginning to 'chatter' at mid to higher throttle values when I combined a sunnysky 2216 650kv with 6S voltage and a 10" prop. Simply switching to a little lower voltage 4S and higher KV 800kv motor worked. Higher PWM rates did help a little bit as well but not enough in the 6s setup, similar to what you are seeing. I think the reality is that 6S with low KV and big props present the biggest challenge for ESC's to read back emf and maintain sync. Is their a significant weight difference between the prop types? For me the lighter props had less issues

Xedos9er, 2020-09-22:
Thanks for the input. This problem appers only running the bi-blade props. The triblades I ran before weight 7.5g, the new ones are even lighter (6g). Motors are perfectly fine, since I always check bearings and also the prop balance is perfect. I even balance them directly on the motor. When looking at gyro traces you notice that there´s literly no noise, but somehow P and D get´s crazy at a specific throttle range. I´m also running fixed timing @23, maybe I should test autotiming??
6S and low kv is pain in ass, I know... 
In fact it´s running perfectly on 3 bladed props.

FelixvdDonk, 2020-09-22:
Yes, auto timing on, 48khz, and sine modulation off gave me the best results. Also don't be tempted to use the RPM filter until the motors run without losing sync. When losing sync the eRPM reported back to the flight controller is probably not reflecting the actual RPM of the motors and therefore the filtering will be ineffective. I think there is a very good chance this is purely an ESC motor/prop combo problem.

Xedos9er, 2020-09-24:
Yesterday one prop exploded right on the hub. I noticed that they don´t seel these props anymore. These were 7x4.5 V1S, new ones are called "durable props 7045" and have a different (POPO compatible) hub. The new ones weight a bit more, seems like they are little bit stronger. I have the feeling that those props just startet to "resonate" on mid throttle.  Hope the new series will fix this issue.

closed 2020-11-26 by DzikuVx