iNavFlight/inav #6148: Unexpected RTH behaviour

Phil-MC, 2020-09-24:
## Current Behavior
On selecting RTH mode quadcopter ascends to required altitude, rotates and heads off in an unrelated direction. It repeats this on another day after a fresh boot and heads in the same direction as before.

## Steps to Reproduce

1. Power up quad with RC transmitter on
2. Wait for HDOP of less than 3
3. Arm in HORIZON mode
4. Switch to NAV ALTHOLD
5. Switch to NAV POSHOLD
6. Fly briefly around
7. Switch to RTH

## Expected behavior
Quadcopter obeys RTH:
Rises to preset altitude
Relocates to GPS coordinates of arming location
Descends and lands within a metre of arming location.

## Suggested solution(s)
How could it 'remember' past a reboot the previous RTH_HOME or whatever it is called? Perhaps it is overwritten by another process. I was unable to find any reference as to how the home coordinates are stored.
## Additional context
The craft has previously obeyed RTH reliably. In NAV POSHOLD mode it flies in a stable fashion and attempts to counteract wind drift. I have a blackbox log of the event but cannot see how to add the file which is not pure text.

Pastebin link to dump [https://pastebin.com/uGv8QcYd](url)

- FC Board name and vendor:
MATEK Systems F722-miniSE
- INAV version string:
version
INAV/MATEKF722MINI 2.5.1 Jul  9 2020 / 12:29:00 ()
GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Available from MATEK download as 'MATEK F722 MINI DPS310' with baro enabled

Phil-MC, 2020-09-25:
I have since re-flashed the firmware and undone the last change (I think) before the event.
`set dynamic_gyro_notch_range = HIGH`
`set dynamic_gyro_notch_range = MEDIUM `

I plan to retest RTH weather permitting...

Phil-MC, 2020-09-25:
[blackbox_log_2020-09-15_155256.TXT](https://github.com/iNavFlight/inav/files/5284641/blackbox_log_2020-09-15_155256.TXT)
Here is the blackbox log for the first flyaway.

stronnag, 2020-09-25:
Doesn't seem to be much evidence of a flyaway, max range 7m.

```
$ bbsummary /home/jrh/dl/blackbox_log_2020-09-15_155256.TXT
Log      : blackbox_log_2020-09-15_155256.TXT / 1
Craft    : TYRO79B on 2020-09-15T14:40:14.901+00:00
Fireware : INAV 2.5.1 () MATEKF722MINI of Jul  9 2020 12:29:00
Size     : 13.67 MB
Altitude : 0.9 m at 00:22
Speed    : 3.1 m/s at 00:43
Range    : 7 m at 01:00
Current  : 64.2 A at 00:00
Distance : 66 m
Duration : 01:41
Disarm   : Switch
```

Nor any RTH attempt:

```
$ inav_modes.rb ~/dl/blackbox_log_2020-09-15_155256.TXT
blackbox_log_2020-09-15_155256.TXT: INAV 2.5.1 () MATEKF722MINI
Iteration	Time(s)	Elapsed(s)	State
        0	  78.6	(   0.0)	 ARM|ANGLE
    11072	  84.3	(   5.6)	 ARM|ANGLE|CAMERA2
    15520	  86.5	(   7.9)	 ARM|NAVALTHOLD|CAMERA2
    15648	  86.6	(   8.0)	 ARM|NAVPOSHOLD|CAMERA2
   198688	 179.8	( 101.2)	 NAVPOSHOLD|CAMERA2
   199177	 180.1	( 101.5)	end of log
```

Phil-MC, 2020-09-26:
Hi stronnagSorry I realised yesterday that the dump I initially pasted was the wrong one and corrected it. The log shows an RTH and BEEP applied when I view it. I will add detail later today.. its too early in the morning at the moment.The version is a custom compile from MATEK for a new flight controller with a change of baro.I plan to put 2.6 on it which has a compile with the new baro. I've seen it in the code.PhilipGet Outlook for AndroidFrom: stronnag <notifications@githubcom>Sent: Friday, 25 September 2020, 19:45To: iNavFlight/inavCc: Philip; AuthorSubject: Re: [iNavFlight/inav] Unexpected RTH behaviour (#6148)
Nor does seem to be a build / revision known to the inav developers:
H Firmware revision:INAV 2.5.1 () MATEKF722MINI

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iNavFlight/inav/issues/6148#issuecomment-699092299",
"url": "https://github.com/iNavFlight/inav/issues/6148#issuecomment-699092299",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Phil-MC, 2020-09-26:
Correct blackbox file to follow later

Phil-MC, 2020-09-26:
[blackbox_log_2020-09-08_185401.zip](https://github.com/iNavFlight/inav/files/5286839/blackbox_log_2020-09-08_185401.zip)
The correct blackbox log zipped with 7-Zip.
RTH attempt at 01:30.336 in section 1/2

stronnag, 2020-09-26:
- Pilot engages RTH
- Craft ascends to 30m
- Craft returns directly to home
- Craft descends to ground

So what's the problem again?
In the replay screenshot, orange icon is home, green track is PH, yellow is RTH.

Please note we have social media channels for support, so far is this just wasting developer's time as there's no evidence of any bug in the logs you've offered to date.
[image]

Phil-MC, 2020-09-26:
Thanks for looking at that, but I think it is the wrong part of the log. I will see if I can give you some evidence tomorrow.

Phil-MC, 2020-09-27:
[image]
This is the flight path from blackbox_log_2020-09-08_185401.zip just above. I have selected the second flight from the log and plotted it. As you can see it ended a long way from where it was armed!
I think this log contains evidence of a bug.

stronnag, 2020-09-27:
No,it merely shows completely that you haven;t verified your hardware, there is essentially zero correlation between the magnetometer reading and the GPS CoG, probably caused by magnetic interference from cabling.

RTH was been working correctly in inav for more than 5 years on properly configured aircraft.

Phil-MC, 2020-09-27:
stronnag  Thanks for your explanation. I will look into it. I also found that are as far as this log is concerned RTH was not implemented. I had BEEPER tied to RTH as I thought I would want both together, but only BEEPER was implemented in this flight. So I must check my modes switches as well.
I thought I was helping by informing about a bug but in fact I wasn't...
Thanks for your help. 
BTW what software did you use to generate your replay picture?

stronnag, 2020-09-27:
You really need to sort out your aircraft; from the log it appears to be somewhat dangerous:

* The compass is obviously not working, random disparity between the compass heading and GPS course over ground; on a well sorted vehicle, these will be within a few degrees;
* Your machine suffers failsafe within 100m of the launch point. Due to the not-working compass it flies away, eventually crashing with a failsafe disarm.

See attached for evidence,images from [mwp's]( https://github.com/stronnag/mwptools) blackbox replay.
[image]
[image]

The wiki contains advice on verifying the compass *before* you rely on it.

closed 2020-09-27 by stronnag

Phil-MC, 2020-09-27:
Thanks for the extra assessment. I am definitely going to check the points you made out. I was putting my latest custom quadcopter through its paces when It went AWL.
Thanks for the link to your mwp tools. It will come in very handy on my ubuntu box. Are you developing a version for Raspberry Pi (armhf)?

stronnag, 2020-09-27:
> Thanks for the extra assessment. I am definitely going to check the points you made out. I was putting my latest custom quadcopter through its paces when It went AWL.
> Thanks for the link to your mwp [...]

It's run on Rpi since mwp was first releases (c. 2014); albeit slowly due to weak GPU in RPi v1. The later versions are probably better; runs OK on my c.2013 ARM chromebook.

Phil-MC, 2020-09-27:
I will try it on a Pi4 laptop I have assembled.

Phil-MC, 2020-09-29:
Runs well on the Pi4.