iNavFlight/inav #6151: Add 3D Home Distance variable

MrD-RC, 2020-09-24:
I would like to see a new 3D Home Distance variable in iNav. It could be used in the OSD and with Logic Conditions. 

It would combine distance from home with absolute altitude (negative altitudes will give a positive addition to the home distance)

This could be used to give a true distance from home and trip distance in 3D space, rather than just along the ground. It could also be used in logical conditions to perform 3D distance based functions.

MrD-RC, 2020-10-17:
It would actually be better if this used some basic trigonometry to calculate the line from the home position to the model, instead of just adding the distance from home to the altitude. This should be simple to do in the flight controller.

For example:
2000m away and 1000m up
In my example would be 3000m
But, the actual 3D home distance should be 2236m

Calculation:
2000 x 2000 = 4000000
1000 x 1000 = 1000000
4000000 + 1000000 = 5000000
sqrt (5000000) = 2236 (rounded to 0 decimal places)

avs2, 2020-10-20:
I think this is addressed by #6226.

MrD-RC, 2020-10-20:
Perfect @avs2 thank you.

MrD-RC, 2020-10-20:
The only thing in the original request that I can’t see (I’m looking on a mobile, so could be me) is the ability to add the 3D home distance to the OSD.

DzikuVx, 2020-10-20:
GVARs can be displayed on the OSD

avs2, 2020-10-20:
I personally would be hesitant to have the 3D distance on the osd instead of the ground distance. For me the use case of this lc is to let the vtx power depend on the 3D distance since that is the distance the signal has to travel. For navigation purposes 3D distance could be rather confusing in the case you are flying directly overhead but you read a large distance on the osd.

closed 2021-11-05 by MrD-RC