iNavFlight/inav #6152: Add support for i2c blackbox

rhass, 2020-09-25:
## Current Behavior
The blackbox currently supports on-board flash, on-board sd-card, and serial. However, it does not currently support i2c. 

## Desired Behavior
I would like to be able to use an i2c data recorder and specify the i2c address of the data recorder. An example of a pre-fab would be the [Qwiic OpenLog](https://www.sparkfun.com/products/15164). This also has the advantage enabling the possibilty to use another inexpensive MCU as an i2c blackbox, such as an ESP8266 which when programmed properly would be possible to download the data via wifi after recovery without having to remove an microsd card.

## Suggested Solution
Extend blackbox_io to support `USE_I2C`.

## Who does this impact? Who is this for?
For smaller flight controllers such as the F411-WING there is not an onboard blackbox, and there often not enough serial ports available to be able to add a serial blackbox. However, i2c is good alternative as it is a shared bus with a reasonably high amount of bandwidth.

DzikuVx, 2020-09-25:
Due to amount of data we log into the blackbox, logging frequency and low bandwidth of the I2C bus, this will not happen.
You have following options:
* built in sdcard
* build in flash
* external, serial connected openlog liek this one https://inavflight.com/shop/s/bg/1461328

closed 2020-09-25 by DzikuVx

rhass, 2020-09-25:
Sorry to beat a dead horse, I just want to better understand the limitations. 

The recommended i2c frequency appears to be 400kHz and 800kHz if the hardware supports it. The max baud rate for a serial blackbox appears to be 250kBps. I understand i2c is a shared bus and has a some amount of overhead. However, I thought the 400kHz i2c connection should be theoretically be about 29% faster than a 250kBps serial connection and 58% faster for 800kHz i2c (accounting for overhead of 1 bit per packet and assuming an unrealistic 0% packet loss). I may be just fundamentally misunderstanding how i2c works by interpreting 1hz == 1bps though.

If 1hz ~= 1bps then I am curious what the limitation is in this case. I understand there is a LOT of data logged, but what I don't understand is the concern about i2c being too slow of a bus. Is the concern that data will be lost or the MCU will exhaust it's memory buffer if other device(s) on the  i2c bus become too chatty, or is there another factor?

Thank you!

rhass, 2020-09-25:
I may have found the answer in the product details for the Qwiic OpenLog  as it "will record data up to 20,000 bytes per second at 400kHz." This is substantially slower than the serial port.

DzikuVx, 2020-09-26:
On top of that, I2C is shared between other sensors, there is huge overhead
for each packet and I2C implementation inav uses is blocking. So, long
story short, not fast enough