iNavFlight/inav #6153: The OSD power level VTX display is still 1 even after arming

romko1, 2020-09-25:
Hello,
I had: FC: MatekF405 mini  and VTX: Rush Tank Mini
Inav v.2.4.0 and the display of the VTX power level in the OSD changes when arming as I set in inav..worked perfectly.
(smartaudio connected to S5 pin and via softserial).

I replaced VTX Rush with VTX TBS Unify PRO32 HV and the VTX power display is still 1.
TBS Unify PRO32 HV supports smartaudio v.2.1. So I installed FW inav v.2.5.2. And I still see a VTX power level of 1 on the OSD. 
I switched the smartaudio connection to the TX4 PIN (I canceled the softserial), unfortunately without success.
It looks like after arming the VTX switches to higher power (is noticeably warmer) but I don't have the courage to send a drone into the air.
Therefore, I have taken the liberty of opening this issue, and I want to ask for advice, an opinion, what should I do?
Thank you.

b14ckyy, 2020-09-26:
Smart Audio should work fine on the F405 Wing FC. please add a dump. What Power level is set in the configuration tab of the inav configurator?

romko1, 2020-09-27:
My dump:
# version
# INAV/MATEKF405 2.6.0 Sep 24 2020 / 19:02:07 (fecc9070)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -RPM_FILTERS
feature -TELEMETRY
feature -CURRENT_METER
feature -REVERSIBLE_MOTORS
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature THR_VBAT_COMP
feature VBAT
feature TX_PROF_SEL
feature BAT_PROF_AUTOSWITCH
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE
feature OSD

# map
map AETR

# serial
serial 20 1 115200 115200 0 115200
serial 0 2 115200 115200 0 115200
serial 1 0 115200 115200 0 115200
serial 2 0 115200 115200 0 115200
serial 3 2048 115200 115200 0 115200
serial 4 64 115200 115200 0 115200

# master
set looptime = 500
set gyro_sync = ON
[... 451 lines cut ...]
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 50
set manual_rc_expo = 1
set manual_rc_yaw_expo = 10
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 430
set vbat_min_cell_voltage = 310
set vbat_warning_cell_voltage = 330
set battery_capacity = 1550
set battery_capacity_warning = 310
set battery_capacity_critical = 232
set battery_capacity_unit = MAH

# end the command batch
batch end

romko1, 2020-09-27:
My dump:
# version
# INAV/MATEKF405 2.6.0 Sep 24 2020 / 19:02:07 (fecc9070)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -RPM_FILTERS
feature -TELEMETRY
feature -CURRENT_METER
feature -REVERSIBLE_MOTORS
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature THR_VBAT_COMP
feature VBAT
feature TX_PROF_SEL
feature BAT_PROF_AUTOSWITCH
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE
feature OSD

# map
map AETR

# serial
serial 20 1 115200 115200 0 115200
serial 0 2 115200 115200 0 115200
serial 1 0 115200 115200 0 115200
serial 2 0 115200 115200 0 115200
serial 3 2048 115200 115200 0 115200
serial 4 64 115200 115200 0 115200

# master
set looptime = 500
set gyro_sync = ON
[... 451 lines cut ...]
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 50
set manual_rc_expo = 1
set manual_rc_yaw_expo = 10
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 430
set vbat_min_cell_voltage = 310
set vbat_warning_cell_voltage = 330
set battery_capacity = 1550
set battery_capacity_warning = 310
set battery_capacity_critical = 232
set battery_capacity_unit = MAH

# end the command batch
batch end

romko1, 2020-09-28:
I just tried ARM with the current settings:
set vtx_band = 4
set vtx_channel = 2
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_pit_mode_chan = 1
set vtx_max_power_override = 4

and it still shows 1.

Thanks for help.

romko1, 2020-10-01:
Hello, 
Please have anyone vtx TBS with Smartaudio v.2.1. 
Could it please test whether the power level in the OSD changes from 1 to another value in ARMING?
Thank you.

romko1, 2020-10-03:
Hello, 
So the problem is fixed. Bad VTX :(.
I replaced the VTX TBS Unify PRO32 HV with a second one, 
and Inavflight (OSD) already shows me different levels of power.
The mistake was that the first VTX, although unlocked, was not actually unlocked :(.

closed 2020-10-03 by romko1