iNavFlight/inav #6154: Is Building in Docker known to work within ubuntu 20.04?

pitts-mo, 2020-09-26:
**Summary:**
I am attempting to add test support for TARGET=KAKUTEF7MINIV3 but even with fresh inav git I keep hitting:
**make: *** No targets specified and no makefile found.  Stop.**

My other STM build dockers appear to be ok... If this is known to work any suggestions are appreciated.
Thanks,
-p

**example:**
user@host:~/inav$ ./build.sh MATEKF405SE
*** Building target MATEKF405SE

make: *** No targets specified and no makefile found.  Stop.

**example with alternate docker image:**
user@host:~/inav$ docker run --rm -v `pwd`:/home/src/ flyandi/docker-inav make TARGET=MATEKF405SE
make: *** No targets specified and no makefile found.  Stop.

**system info**:
uname -a
Linux h170m 5.4.0-48-generic #52-Ubuntu SMP Thu Sep 10 10:58:49 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.1 LTS
Release:        20.04
Codename:       focal

docker --version
Docker version 19.03.13, build 4484c46d9d

pitts-mo, 2020-09-26:
I get the same in a fresh 18.04.5 install. Am I hitting a classic PEBKAC error?
Thanks,
-p

example:
fresh 18.04.5 amd64 VirtualBox after installing docker via https://docs.docker.com/engine/install/ubuntu/#install-using-the-repository

lubuntu@lubuntu-VirtualBox:~$ git clone https://github.com/iNavFlight/inav
lubuntu@lubuntu-VirtualBox:~$ cd inav/
lubuntu@lubuntu-VirtualBox:~/inav$ ./build.sh MATEKF405SE #showing result of second run
*** Building target MATEKF405SE

make: *** No targets specified and no makefile found.  Stop.

stronnag, 2020-09-26:
Please see the latest development documentation. `cmake` is now used for building.

pitts-mo, 2020-09-26:
Thank you @stronnag 
This did the trick:
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Linux.md

I have preferred tinkering in docker based build environments ... maybe if necessity exists and someone has time cmake and the initial cmake task might end up in a docker image.

Thank you very much.
-p

closed 2020-09-26 by pitts-mo