iNavFlight/inav #6155: [2.6 (c6cfff47)] parsing error when adjrange for TPA and TPA BP is set via CLI

b14ckyy, 2020-09-26:
## Current Behavior
When I set adjrange settings for TPA Breakpoint or TPA Factor in CLI, it outputs a parsing error and resets the existing value fo the adjustment slot in INAV. Setting TPA adjustments via Configurator GUI works fine and is stored correctly. 
Reading out the adjrange shows valid values. Copying the output and send them back via CLI results in a parsing error and reset of the slots. 
[image]

## Steps to Reproduce
1. set an adjustment for TPA and TPA breakpoint via configurator GUI
2. read adjrange via CLI > valid output
3. copy the adjrange lines and load them back into CLI
>> Lines with Item ID 50 and 51 will cause a parsing error

Issue also reproduced with release version of configurator 2.5.2 (using snapshot build to have compatibility with 2.6 snapshot) in CLI only mode and also with betaflight configurator. Same result. All other adjrange slots are restored just fine. 

## Expected behavior
loading back adjrange settings without error in case of backup restore

## Suggested solution(s)
fix in INAV

## Additional context
Dump: https://pastebin.com/09Fy9uas

Excample Lines in my case: 
adjrange 0 0 4 1025 1075 18 5
adjrange 1 0 4 1075 1125 19 5
adjrange 2 0 4 1125 1175 20 5
adjrange 3 0 4 1175 1225 15 5
adjrange 4 0 4 1225 1275 16 5
adjrange 5 0 4 1275 1325 17 5
adjrange 6 0 4 1325 1375 9 5
adjrange 7 0 4 1375 1425 10 5
adjrange 8 0 4 1425 1475 11 5
adjrange 9 0 4 1475 1525 28 5
adjrange 10 0 4 1525 1575 29 5
adjrange 11 0 4 1575 1625 31 5
adjrange 12 0 4 1625 1675 48 5
adjrange 13 0 4 1675 1725 50 5
adjrange 14 0 4 1725 1775 51 5
adjrange 15 0 4 1775 1825 32 5
adjrange 16 0 4 1825 1875 35 5
adjrange 17 0 4 1875 1925 36 5
adjrange 18 0 4 1925 1975 37 5
adjrange 19 0 4 1975 2025 0 5
---

- FC Board name and vendor: Omnibus F4 V6
- INAV version string:
INAV/OMNIBUSF4V6 2.6.0 Sep 26 2020 / 12:56:06 (c6cfff47)
GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

b14ckyy, 2020-09-26:
Update: The TPA Adjustments are also not shown in the OSD. There is only a 3-bar RSSI icon followed by a 0. So something seems to be mapped incorrectly.

Scavanger, 2020-10-28:
Fixed in #6242

closed 2020-12-11 by DzikuVx