iNavFlight/inav #6157: althold and poshold problem

gerdal91, 2020-09-27:
Hello I came from betaflight . because I want to use nav pos hold and alt hold but I couldt get it work.

when I turn on ALTHOLD or POSHOLD or BOTH together. my quad goes up very quickly then starts sink until floor.
I checked magnotometer its work good.
Can someone give me a solution?
[LOG00018.TXT](https://github.com/iNavFlight/inav/files/5288570/LOG00018.TXT)
[LOG00016.TXT](https://github.com/iNavFlight/inav/files/5288572/LOG00016.TXT)
[LOG00014.TXT](https://github.com/iNavFlight/inav/files/5288573/LOG00014.TXT)
[LOG00012.TXT](https://github.com/iNavFlight/inav/files/5288574/LOG00012.TXT)

`dump

# version
# INAV/MATEKF405 2.5.2 Aug  4 2020 / 10:15:56 (faaedc7cb)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# logic
logic 0 0 -1 0 0 0 0 0 0
logic 1 0 -1 0 0 0 0 0 0
logic 2 0 -1 0 0 0 0 0 0
logic 3 0 -1 0 0 0 0 0 0
logic 4 0 -1 0 0 0 0 0 0
logic 5 0 -1 0 0 0 0 0 0
logic 6 0 -1 0 0 0 0 0 0
logic 7 0 -1 0 0 0 0 0 0
logic 8 0 -1 0 0 0 0 0 0
logic 9 0 -1 0 0 0 0 0 0
logic 10 0 -1 0 0 0 0 0 0
logic 11 0 -1 0 0 0 0 0 0
logic 12 0 -1 0 0 0 0 0 0
logic 13 0 -1 0 0 0 0 0 0
logic 14 0 -1 0 0 0 0 0 0
logic 15 0 -1 0 0 0 0 0 0

# gvar
[... 1168 lines cut ...]
set yaw_rate = 60
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 4
set vbat_cell_detect_voltage = 419
set vbat_max_cell_voltage = 400
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# end the command batch
batch end

# `

MartianLocust, 2020-09-29:
Have you placed some open celled foam over the baro?  I had the same behavior, then added foam and its a steady as a rock now. The baros are very sensitive and any airflow across the sensor will cause pressure fluctuations.

gerdal91, 2020-10-03:
> Have you placed some open celled foam over the baro? I had the same behavior, then added foam and its a steady as a rock now. The baros are very sensitive and any airflow across the sensor will cause pressure fluctuations.
I tried today but same

Beardyblair, 2020-10-11:
have you set the throttle mid value to match your craft? (my guess is no based on values above)
Read the manual for setup and this should help 👍 https://github.com/iNavFlight/inav/wiki/Navigation-modes

Mim0oo, 2020-10-16:
You have to configure your hover throttle or as Beardyblair commented your mid throttle if you have enabled mid throttle for hovering.
Second advise, very important: Do not disable air mode by any means. Keep it always active otherwise makes dron to rocket jump unexpectedly.

gerdal91, 2020-10-17:
thank you every one the problem was the barometer. the barometer increasing continuously.

A2BELE, 2020-10-23:
Mine does same thing. Matek F722se.
Shows 0m sitting on the ground but as it climbs it goes into the negative so of course when I try to altitude hold it shoots you in the air. 
Was it a confirmed fault on the board?

gerdal91, 2020-10-23:
> Mine does same thing. Matek F722se.
> Shows 0m sitting on the ground but as it climbs it goes into the negative so of course when I try to altitude hold it shoots you in the air.
> Was it a confirmed fault on the board?

Connect to computer in sensor tab there is barometer check if the barometer works correctly. It should increase when you move up and lover when you move down .

MartianLocust, 2020-10-23:
Mine is Matek F722 miniSE and it works as intended. I used the Matek firmware to get the DPS310 baro working. Had a blip yesterday, but that was because the foam over the baro had shifted.

cjmv, 2020-12-26:
Mine is a Matek F722 miniSE together with the a Matek M8Q-5883 (GPS and compass). I think it's better after covering the baro with the foam as suggested, but it sets the altitude to -3, -5 m after arming (before the foam it was something -15m). The value of the Baro for me was always quite stable. The Altitude value, which I think it's a calculated value, something to do with the AHRS - Altitude Heading Reference System, has never been stable.

I do have a spare Matek F722 miniSE, but it hurts just to think on all the prep needed for me to install the board and solder everything once again before exhausting all other possibilities. :-)

The other thing is... I don't know if any of you is using Matek M8Q-5883, but whenever I look into the blackbox logs, the compass values only show X and Y axis. The Z axis is a flat line.

Any further thoughts about this?

closed 2021-05-05 by DzikuVx