iNavFlight/inav #6159: BN 220 suddenly stopped working

Strat00s, 2020-09-27:
First time using iNav. Built my setup and everything worked as expected (tested it multiple times). And today, suddenly after kinda nothing (just powering it on and going through some settings) it started beeping like crazy. After checking the status (in the console) it showed that the GPS is not working. Which is pretty strange as nothing happened to the system. So I tested it with my BN 880 and that one works without any problems.
Trying to remove some weird software bug. I reflashed the board and tested it again. Got the same outcome. Changing baude rate also did absolutely nothing. So my BN 220 just died, right? Wrong.

Flashing ardupilot onto the board I get the GPS working without any problems (both the 220 and 880).
So, do you guys have any idea, what the problem might be?
I doubt that posting any config would help you guys, as I just flash the board and boot it and enable the gps.

DzikuVx, 2020-09-28:
try lowering baud rate on the serial port to 38k

Strat00s, 2020-09-28:
> try lowering baud rate on the serial port to 38k

Tried every baud rate and every serial port (1, 3, 6).
I read somewhere, that this might be caused by some leftover stuff, but I always select full chip erase :/

rsilk1949, 2020-09-30:
Double check your wiring and connections.

Strat00s, 2020-09-30:
> Double check your wiring and connections.

Kinda obvious, that's not the problem. And I checked multiple times. Making sure that there is continuity between everything.

Strat00s, 2020-10-02:
Bump. 
Seriously guys? Nobody as the slightest idea, why it works in ardupilot, but suddenly doesn't in iNav?

digitalentity, 2020-10-03:
A bug is something that doesn't work from the start and is reproducible by multiple users. If this was a bug, we'll have reports all over the place as this GPS module is very popular. In your case a spontaneous malfunction most likely indicate a hardware issue.

Please double-check your connections as suggested before, provide a dump of your settings and a photo of how your module is connected. Then we might be able to identify a possible issue.

Strat00s, 2020-10-03:
> A bug is something that doesn't work from the start and is reproducible by multiple users. If this was a bug, we'll have reports all over the place as this GPS module is very popular. In your case a [...]

If you read my response, I even did a continuity check to be 100% sure, that the wiring is correct and there is a connection. Also, going from ardupilot to iNav, without any hardware changes (only flashing the different FW) makes the GPOS not work anymore.
If you think that it is a spontaneous malfunction in 2 different modules at the same time, sure. But I highly doubt that as they work in ardupilot.
In the picture,

[image]
[profile0_dump.txt](https://github.com/iNavFlight/inav/files/5321346/profile0_dump.txt)

there  is just the GPS connected to UART6 and the FC (and rest of my non working plane, which does not really matter now). Just putting ardupilot on it (not changing anything else) gives me GPS immediately. iNav just screeches to hell, that it's not there or that it can't communicate or whatever.

digitalentity, 2020-10-03:
Possible troubleshooting steps (do incrementally)

1. Switch autoconfig off (`set gps_auto_config = OFF`)
1. Try switching to NMEA protocol and keep the auto-baud on. Note that it may take a while to get detected.
1. Roll back to INAV 2.0 or earlier and configure for UBLOX.

If any of these will get GPS live - there is a problem with `FC_TX -> GPS_RX` connectivity. 

If neither of the steps will get the GPS live - this is something odd happening and using a logic probe to sniff the traffic between FC and GPS would be necessary to figure out what's going on.

Strat00s, 2020-10-03:
> Possible troubleshooting steps (do incrementally)
> 
> 1. Switch autoconfig off (`set gps_auto_config = OFF`)
> 2. Try switching to NMEA protocol and keep the auto-baud on. Note that it may take a while to get detected.
> 3. Roll back to INAV 2.0 or earlier and configure for UBLOX.
> 
> If any of these will get GPS live - there is a problem with `FC_TX -> GPS_RX` connectivity.
> 
> If neither of the steps will get the GPS live - this is something odd happening and using a logic probe to sniff the traffic between FC and GPS would be necessary to figure out what's going on.

1. nothing changed
2. Nothing changed (gps_auto_baud = ON)
3. Always used the 2.5.2 (just to make that clear). 2.0 did not change anything.

Also, do you have any idea, why it did work before and it's not working now? That just goes against anything you are suggesting. (because it freaking worked)

Strat00s, 2020-10-05:
I guess that I am just going to use ardupilot 🤷

digitalentity, 2020-10-05:
The symptoms so far suggest a hardware issue with either the GPS module or the FC, especially given the fact that it worked before on the same setup. Replacing components to known good parts will help isolating what's wrong. There is a very little possibility that this is a software issue, but it will require sniffing the traffic on the wire with a logic probe and decoding what GPS is sending and why it doesn't respond to the commands from the FC.

The fact that ardupilot "detects" the GPS doesn't mean that it's going to be usable for real-time airplane navigation. My bet is that TX->RX line is broken and GPS is falling back to 1Hz NMEA @ 9600 baud which is way too little for navigation, but still enough to see GPS as "alive" by some firmwares.

I recommend avoiding that GPS or that FC regardless of the firmware until you know why this is happening.

Strat00s, 2020-10-05:
> The symptoms so far suggest a hardware issue with either the GPS module or the FC, especially given the fact that it worked before on the same setup. Replacing components to known good parts will [...]

Ardupilot detects them both as ublox (if I understand this correctly)
BN-880:
```
05.10.2020 15:30:58 : u-blox 1 HW: 00080000 SW: ROM CORE 3.01 (107888)
05.10.2020 15:30:56 : GPS: u-blox 1 saving config
05.10.2020 15:30:42 : GPS 1: detected as u-blox at 230400 baud
```
BN-220:
```
05.10.2020 15:33:12 : u-blox 1 HW: 00080000 SW: 2.01 (75331)
05.10.2020 15:33:10 : GPS: u-blox 1 saving config
05.10.2020 15:32:56 : GPS 1: detected as u-blox at 230400 baud
```

digitalentity, 2020-10-05:
That's something already, many thanks!

I see they somehow ended up being configured for 230400 baud (by ardupilot probably). This config got saved on the module, therefore INAV is unable to communicate to the device (it supports only up to 115200 baud).

Changing this to the bug, thanks for being persistent :smile:

digitalentity, 2020-10-05:
Can you please test this build? It's based on `2.6.0` so you'll need a new configurator as well.

In this version the GPS autobaud will consider the 230400 baud as well, please try enabling `gps_auto_baud = ON` and configuring the port for 115200.

[inav_2.6.0_OMNIBUSF4PRO_LEDSTRIPM5.zip](https://github.com/iNavFlight/inav/files/5328195/inav_2.6.0_OMNIBUSF4PRO_LEDSTRIPM5.zip)

Strat00s, 2020-10-05:
> Can you please test this build? It's based on `2.6.0` so you'll need a new configurator as well.
>
> In this version the GPS autobaud will consider the 230400 baud as well, please try enabling [...]

It immediately shows up (blue icon on top) and status return GPS=OK. Removing it while powered on, it starts beeping again (seems to be quieter now) and status returns GPS=UNAVAILABLE.
Both seem to work now 👍

digitalentity, 2020-10-05:
Awesome. I'll shape this into a proper fix now.

PR #6182 (2020-10-06): [GPS] Support GPS baud 230400
In some scenarios GPS may end up being configured to 230400 baud (from the factory or by 3-rd party software (Ardupilot, PX4 etc). In this scenario INAV will fail to detect the GPS or reconfigure it to one of the supported baud rates.

This PR adds 230400 to supported baud rates and allows INAV to reconfigure such the GPS modules correctly.

Fixes #6159
closed 2020-10-06 by digitalentity