iNavFlight/inav #6161: RGB LEDs stop working with DSHOT

jaroslawkajoch, 2020-09-28:
## Current Behavior

RGB LEDs stop working with DSHOT.

## Steps to Reproduce

1. Run option to  flashing LEDs always.
2. Change ESC protocol to DSHOT

## Expected behavior

I expect LEDs will still working with all ESC protocols.

Latest version 2.5.2

- FC Board name and vendor: Mamba F405 mini mk2
- INAV version string: 2.5.0. ( latest  version chrome app)

sergeyprokofev, 2020-10-18:
Mamba F405 RGB, LEDs stop working with DSHOT.
Works with MULTISHOT and ONESHOT.

minkyu830, 2020-11-07:
I am experiencing the same thing.
FC : MAMBA F405 MK2
ESC : Mamba F40 ESC (DSHOT 600)
BARO : MS5611
GPS : Beitian BN-880Q GPS (MAG : QMC5883)
FRSKY R-XSR sbus connect

I tested INAV V2.5.2, V2.5.0, V2.4.0, V2.3.0 .
all versions choose dshot protocol, the Multi-color RGB LED strip doesn't work.
I tested it on betaflight and it worked.
I compared the resources of inav and betaflight, the led strip resources is the same A00.
[image]
It looks like inav's own bug, but it doesn't seem to be fixed for over a year.
Can it be fixed or can someone else be in charge of fixing it?
Or is there a workaround way to use dshot and led together?

ipleten, 2020-11-28:
I faced the same issue.

kticka, 2021-05-22:
Same issue. Led strip with DSHOT protocol doesn't work.

wx4cb, 2021-05-22:
must be board specific, i have dshot escs on multiple buildswith different f405 based boards (f405 wse, f405 wing, quad f405 - an off brand one) and they all have LEDS in them, in fact my nano talon 250g with a matek f405 wing has 35 leds on it in the fuse without a problem.

kticka, 2021-05-22:
Have two mambas (F405 MK2), doesn't work on both when DSHOT protocol is enabled.

closed 2022-02-10 by DzikuVx