iNavFlight/inav #6164: Support needed: Mavlink telemetry failed ?

lipk888, 2020-10-01:
Hi team:
TBS udpate crossfire TX V2 and it have in-build  wifi module and wifi can output mavlink telemetry.
i just want to build a ESP32 ATT base on zs6buj's project(AntTracker)

mavlink massage transmit follow below:
INAV mavlink telemetry(F405WING)       -->     crossfire nano reciver   -->    crossfire TX V2 wifi -->          ESP32 ATT
i set TX1/RX1 for mavlink telemetry out               set CH3 to rx                          UDP port:8800            remot UDP port:8800

so now i get heartbeat good feedback from ATT i asuume it means ATT receive the mavlink massage. actully no GPS massage receive @ATT

so i want to know if any way to know whether INAV send out mavlink massage from TX1?
i just upload log file for ATT.
[log.txt](https://github.com/iNavFlight/inav/files/5313365/log.txt)

KingKone, 2020-10-06:
Hi lipk888,
do you have mavlink over wifi working yet? 
I don't know but I don't get it to work with the tango 2.
Looks like you get some data from INAV?

ShikOfTheRa, 2020-10-07:
Out of interest, do you know if xfire config options only supports UDP? I should update to latest to see, but its running so well I'm a little reluctant :)

TBH, I have found the whole wifi / bluetooth connectivity pretty unreliable. I've tried with others and as soon as fly with other users running 2.4g tx or VTX it keeps bombing out. Also seen this with others using at FPV meets.

lipk888, 2020-10-10:
> Hi lipk888,
> do you have mavlink over wifi working yet?
> I don't know but I don't get it to work with the tango 2.
> Looks like you get some data from INAV?

YES, ATT only got mavlink heartbeat massage, but no GPS massage back

lipk888, 2020-10-10:
> Out of interest, do you know if xfire config options only supports UDP? I should update to latest to see, but its running so well I'm a little reluctant :)
>
> TBH, I have found the whole wifi / [...]

well i check the manual current TBS WIFI only can support UDP mavlink output.

closed 2024-03-20 by DzikuVx