iNavFlight/inav #6173: Land after RTH - delayed landing.

PR31D, 2020-10-04:
## Current Behavior
As expected. On the craft returning home it immediately begins to land.

## Desired Behavior
On the craft returning to home, if it could loiter a set amount of time before starting to land. 

## Suggested Solution
Add in a variable delay where the craft will loiter before starting to land.

## Who does this impact? Who is this for?
This is mainly for fixed wing that dont crash well. I fly lion long range and I cant have my failsafe set to RTH and loiter as it could be up there loitering for a long time. However if im at short range and I've an issue with the link, the wing will start to land immediately on return to home. Not giving me any time to try sort the link out. 

## Additional context

PR31D, 2020-10-04:
👎

MrD-RC, 2020-10-26:
This is an excellent idea for pilots who fly from a site unsuitable for auto landing. It will give time to try and fix problems, knowing that the model is safe above them.

PR31D, 2020-10-27:
Hi Darren,

Thanks for the feedback.

MrMap, 2020-11-25:
Very good that this is merged into new versions.
Here is a video illustrating a CFIT ending of a test flight otherwise sussesful (ending with a FS test).
On a side note my fixedwing "landed" on cruise throttle, despite _set failsafe_throttle = 1000 (I will submit an issue on this).

Had this new landing delay functionality been in place I would not have to use so much glue.
That said I am painfully aware that this was all on me, not having thought it through properly (FS test procedure).
Enjoy the video!
https://youtu.be/9HkOczusHDI

b14ckyy, 2024-03-20:
Should be no issue anymore with https://github.com/iNavFlight/inav/pull/9713
Additional features are planned for the near future, for example only autoland with defined landing spots but to avoid circle landing. Will keep this open for now.

b14ckyy, 2024-03-21:
Oh here we go, its implemented. Closing https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#nav_rth_fs_landing_delay

closed 2024-03-21 by b14ckyy